grblHAL/core #364: Maintaining the Z axis after backlash compensation

Melkiyby, 2023-09-13:
In my CNC, the Z axis slides on Teflon pads. If, after moving the Z axis up, it moves down, then over time it slides down by 0.14 mm. This causes different milling depths and ultimately poor machining quality. 
Please add option for Z axis:
If Z makes a downward movement, then the stepper motor returns in the opposite direction by the backlash distance, then if Z moves downward, then the GRBL must add backlash to the movement and return back to the backlash distance, and if Z moves up, then there is no backlash compensation, since the axis Z is already supported by backlash compensation.

terjeio, 2023-09-13:
Backlash correction is [inserted here](https://github.com/grblHAL/core/blob/049f0f13e45211237f3b096083afb8d371aad9fb/motion_control.c#L112-L161), can you try to modify the code to suit your needs?

vkter, 2023-09-20:
Is the Backlash Compensation function now supported?
How to configure the corresponding value for each axis?

terjeio, 2023-09-20:
> Is the Backlash Compensation function now supported?

It is a [compile time option](https://github.com/grblHAL/core/blob/742eed9254faa5a80ab99e7a20db8882683c150b/config.h#L489-L491).

> How to configure the corresponding value for each axis?

$160, $161, $162, ... So in a similar way as other axis settings.

closed 2023-12-15 by terjeio