grblHAL/core #371: Home position of rotational axis A

Avara7, 2023-09-19:
I'm making 4 axis tangential knife CNC CoreXY machine with BTT-SKR 2.0 controller (STM32F407) and I have several issues.
1. The hard limit switch of rotational axis is proximity switch with physical dimension 10x10 mm and his position is not adjustable, located at 0 degree. After homing the axis is taking position 5 or -5 degree (depends of direction rotation during homing) because of switch is triggering and it's normal. But the range of position from -5 to 5 degree is unreachable and it's not possible to put axis to 0 degree or pass limit switch during rotation.  When I am trying implement g-code I am getting alarm "Hard limit is triggered". The setting is $133=1440, $376=1, $376=8 and $376=0.

2. On the "config.h" I'm setting
 #define DEFAULT_CONTROL_SIGNALS_INVERT_MASK   7, 
 #define DEFAULT_AXIS_ROTATIONAL_MASK    8
after downloading to the controller,  GRBLGRU terminal displaying them  as $14=1 and $376=1, even after $RST=* command.
When I'm sending command $376=8 in GRBLGRU terminal, there is message- error 52, unknown error.

3. The setting is $4=7. After power up controller the level of step of enable signal is HIGH. Jogging axis, motor running and after stopping the level of step enable signal is keeping LOW. Only after sending command $RST=* level of step enable signal became HIGH and after machine working well. After power up there is the same issue and I need to send $RST=* again.

4. Sometimes in GRBLGRU interface in the window displaying CNC machine, the parts of machine is not moving (not updating) and value in actual work position not updating too  during jogging or implementing gcode.

 Can you please help me with solving that issues?

[ver:1.1f.20230228:]
[opt:vnmslw,35,1024,4,0]
[axs:4:xyza]
[newopt:enums,rt+,tc,sed]
[firmware:grblhal]
[nvs storage:*flash]
[driver:stm32f407]
[driver version:230227]

GRBLGRU ver 5.1.0

terjeio, 2023-09-19:
1. Hard limits should be disabled for rotational axes? Or better allow per axis enable?
2. What is [ioSender](https://github.com/terjeio/ioSender) displaying? It could be that grblGRU misinterprets settings. $376 uses an axismask for the default value in config.h but shifts this value 3 positions to the right when reporting/setting it. So the default value of 8 becomes 1 when queried/changed via $376 - this because XYZ axes cannot be set as rotational.
3.  I am not able to replicate this. Post all your settings before resetting so I can check.
4. Randomly during a session? Setting $10 might affect this. Does it happen with other senders, e.g ioSender?

BTW the 20230228 build is outdated, you should update to a [newer build](https://github.com/grblHAL/core/blob/master/changelog.md). At least 3. could be due to a bug that has been fixed.

Avara7, 2023-09-20:
Nice, thanks for the tips. I'm going to update GrblHAL and do some more tests.