grblHAL/core #373: Feature request: Program resume support and support for starting program from midway

karoria, 2023-09-26:
Hi @terjeio 
You might have already given a thought for this. I know this is a difficult task to add a resume support at controller level but I would really like if it can be developed and it will be a huge achievement as I don't know any opensource cnc controller with this feature. Especially this feature is useful when power goes off at mid-job or a broken tool when a large program is running. The flow may be like:
1. A program line counter stores latest line number which was just executed.
2. Upon power cycle, controller can give a state message that the machine was in the middle of a job when power was down anf if user wants to resule the job.
3. Upon confirmation, controller reads the program from the first line for accessories and model states till the last line executed.
4. It turns on the accessories and applies modal states.
5. Machine goes to X and Y position of just the previous immediate location.
6. Machine lowers Z position to just the previous immediate Z location.
7. Resumes the program from the line number.

terjeio, 2023-09-26:
Not easy for sure, I'll leave it as a challenge for somebody to write a plugin for it...

The most challenging part is where to _continuously_ store the needed data. Ideally it should be in memory mapped battery backed RAM, which can be in the main RAM if the controller itself is running off an UPS. Forget about EEPROM or flash as these are both slow and would wear out quickly. FRAM _could_ be an option.

There are other challenges as well - protocol extensions, sender updates, ...

karoria, 2023-09-26:
Ok. Do you think that resume support can be achieved by webui instead of sender? What if I want to develop it in my grblTouch webui? I know, I will need some information from controller in that case like which line number is being executed and store it in browser local memory. Again, some commands to be introduced like M**  Ln1234 to start from line 1234. Do you think, some part on controller side and some part on browser side will give a good combo to achieve this?

terjeio, 2023-09-26:
If streaming gcode from a SD card then it becomes a bit easier as the filename and position in the file can be recorded instead of a line number that may or may not be available and has no meaning unless the file it relates to is known.
FYI I[ record the file position](https://github.com/grblHAL/core/blob/70ffca3c0afcddcbf92be837b47cde83295fc979/ngc_flowctrl.c#L264) and rewind to that in the code that implements looping, so that is possible.

> Do you think, some part on controller side and some part on browser side will give a good combo to achieve this?

Possibly, but it will still be a lot of work - mainly for verifying the code?

BTW which line number is executing is a bit problematic as well since motions are queued, the gcode has to have line numbers to track that. And then moves might be partially completed so you will have to rewind to a safe spot before restarting.

hanke-cnc, 2023-09-26:
> If streaming gcode from a SD card then it becomes a bit easier as the filename and position in the file can be recorded instead of a line number that may or may not be available and has no meaning [...]

> If streaming gcode from a SD card then it becomes a bit easier as the filename and position in the file can be recorded instead of a line number that may or may not be available and has no meaning [...]

"I've examined the practices of other Mach3 users, and it appears that they only execute a few lines or a few dozen lines of code ahead of the current execution. This seems quite risky. Of course, I've never tried Mach3 myself."

karoria, 2023-09-26:
We can use line number sent to planner queue instead of line number executed to make it simple. Further some 20-30 lines ahead of it for resuming the program to be on safe side?
When a program is reset midway, grblHAL gives message "Reset program file at line number ****", how do you get that line number?

terjeio, 2023-09-26:
> We can use line number sent to planner queue instead of line number executed to make it simple.

If there is no line number in the gcode none (0) is sent to the planner. Most of the gcode I have seen does not contain line numbers so that is normal practice nowadays?

> When a program is reset midway, grblHAL gives message "Reset program file at line number ****", how do you get that line number?

By [counting number of end-of-line character sequences](https://github.com/grblHAL/Plugin_SD_card/blob/6fd807ef8efdea33533116669e3d953b98e5012c/sdcard.c#L392-L393) encountered. Again, if there are no line numbers in the gcode then 0 is sent to the planner - and I do not parse them out in the SD card code if they happens to be present. IMO it is better, and likely faster to resume, if you record the file position returned by `ftell()`.

cbaugher, 2023-09-27:
Something else to be aware of when restarting a program after power failure, is that depending on how your machine determines its home position, it may not home to *exactly* the same place it was before.  On most hobby machines it's probably not much of a concern, but on some machines it could be.  And if you happen to fail during a tapping or threading operation, the odds of being able to successfully resume there are pretty low.

C|

closed 2026-02-27 by terjeio