grblHAL/core #374: Custom Sender for Multiple Boards on One Serial Line

fireup924, 2023-09-27:
I want to write a custom Sender that can communicate with multiple boards on the same serial port and would like to use half-duplex RS485 so I can connect multiple boards together on one bus. Would this be just a matter of setting the RS485 to receive mode after sending a "?" command to get the status? I know that I have to make each board have a unique ID and only except the commands if the ID matches. My question is if this is possible or do you think there will be complications with timing of when it's sending back status information?

terjeio, 2023-09-27:
A single bus would require a rewrite of the protocol (implemented at the UART handling level - in serial.c?)  where all commands and gcode has the target board ID somehow added to them (using a message container?). Perhaps you would need a collision detection mechanism for responses as well, unless your sender waits for a response from the adressed controller before sending new commands.
Be aware that there is no guarantee that real time report requests will always be honoured 1:1, multiple request _may_ lead to a single response, at least during homing since the homing routine blocks sending out responses.

It might be easier to add a cheap MCU, e.g. a PI Pico, in front of each controller and let that handle the protocol extension rather than changing the code in the controller itself...

fireup924, 2023-09-27:
What do you think about CAN bus? It can send and receive at the same time and it has built-in collision detection. I know that I have to break up a GCode line into multiple packets.

> It might be easier to add a cheap MCU, e.g. a PI Pico, in front of each controller and let that handle the protocol extension rather than changing the code in the controller itself...

I might go this route if I can't get it working with one MCU

terjeio, 2023-09-28:
A bit late last night when I answered...

You can add a plugin that does [protocol decoding by claiming the serial input and decoding the messages it into a new input buffer](https://github.com/grblHAL/Plugins_laser/blob/master/lb_clusters.c) that the core reads from. You will still need to handle potential collisions, depending on what you are trying to achieve with this multi controller setup this could somehow be done in the sender?

CAN bus can be used but needs driver code and a controller board that can drive it, @dresco is working on adding code for that. Some supported processors has inbuild CAN peripherhals that will make it simpler.
But what about ethernet? Could that be an option?

closed 2024-04-16 by terjeio