grblHAL/core #382: cmake includes plugins by default

wiersmapa, 2023-10-23:
Tried to build RP2040 from the git repository and ran into errors (for example in 'unused' trinomatic plugin.)  The CMakeLists.txt file has if(AddMyPlugin) with the list of plugins and and an else() which lists them all again.  I removed the unused plugins from that else() list and it compiled with no errors.

terjeio, 2023-10-24:
Which errors? I do not get any.

wiersmapa, 2023-10-24:
I pulled a new copy and this time it compiled without errors.  I remember a
number of errors related to the TRINOMATIC variable declarations (the
stream variables in tmc_uart.c) which made me think it was related to the
plugins.  Not sure what was different before.  Thanks, Paul

terjeio, 2023-10-26:
Perhaps a bug in the previous version? I stumble across some now and then, mostly minor and annoying ones. Anyway, thanks for reporting.

closed 2023-10-26 by terjeio