grblHAL/core #383: Is there away to add implement a "map", where a pwm of 0=0, the smallest increment above 0 detectable by micro can be set to a selected pwm value, and what a program may perceive as 100% to another value(75% duty). I know that now you can do the later and the available option on lower value is almost the oposite of what im looking for..... also itd be nice to be tunable with out recompiling. Using the $$ settings or potentiometer would be slicker. Thanks

terjeio, 2023-10-24:
Is there away to add implement a "map", where a pwm of 0=0, the smallest increment above 0 detectable by micro can be set to a selected pwm value, and what a program may perceive as 100% to another value(75% duty). I know that now you can do the later and the available option on lower value is almost the oposite of what im looking for..... also itd be nice to be tunable with out recompiling. Using the $$ settings or potentiometer would be slicker. Thanks

_Originally posted by @customsolutions in https://github.com/grblHAL/core/discussions/203#discussioncomment-7369539_

terjeio, 2023-10-24:
I am not sure what you mean by this. $34-36 does not handle what you are after?

[image]

A plugin can override spindle functions or implement custom spindle drivers like [this](https://github.com/grblHAL/Templates/blob/master/my_plugin/Solenoid_spindle/my_plugin.c) or [these](https://github.com/grblHAL/Plugins_spindle/). So you are not stuck with what the standard implementation provides.

phil-barrett, 2023-10-24:
I think what he is asking is for a PWM mapping function. In general terms if would be f(PWM_in) = PWM_out.  

In practical terms it could be a table of PWM with linear interpolation between points.  For example 11 entries in steps of 10%
fPWM[0] = 0.0  // 0%
fPWM[1] = 0.08  // 10%
fPWM[2] = 0.16  // 20%
fPWM[3] = 0.24  // 30%
fPWM[4] = 0.32  // 40%
fPWM[5] = 0.40  // 50%
fPWM[6] = 0.47  // 60%
fPWM[7] = 0.53  // 70%
fPWM[8] = 0.61  // 80%
fPWM[9] = 0.68  // 90%
fPWM[10] = 0.75  // 100%

So a requested PWM of 50% would actually be 40% PWM output. A requested PWM of 55% would be 43.5% ( (40+47)/2 ).

This could be useful in linearizing PWM and 0-10V output. A simple PWM to 0-10V converter often has some "sag" in the mid range and could benefit from a "boost" mapping. Also, I've noticed that straight PWM input on several HY VFDs I have is not very linear - sometimes the mid range is quite a bit low.  They could benefit from a boost mapping. 

It doesn't need to be steps of 10%.  Could be a 5 entry table for steps of 25%.  Also, allowing greater than 100% on the output would be useful.

terjeio, 2023-10-24:
> I think what he is asking is for a PWM mapping function.

Something like [this](https://github.com/grblHAL/core/wiki/Spindle-sync#spindle-rpm-linearization) then?

phil-barrett, 2023-10-24:
Yes.  I had seen that but presumed it was only for lathe mode. It looks like a possibility but I don't quite understand how to use it. Needs a grblHAL build with spindle sync and tuning options, I presume. Will check it out today. My test bed for Pro is on a shelf and I have 100 Pico ETH boards to do intake processing on so will be later today.  I'm sure I will have questions...

terjeio, 2023-10-24:
> Yes. I had seen that but presumed it was only for lathe mode.

It is not, it only makes it possible to automate data collection since real RPM is available from the controller. One can use a tachometer to collect data points for the python algorithm.

Settings $66 - $69 becomes available when [spindle linearization](https://github.com/grblHAL/core/blob/04a5edcf8353f7294167d12d32e4b073d9be9ce4/config.h#L93-L103) is enabled, it is not dependent on lathe mode. I guess I should make this option available from the Web Builder.

customsolutions, 2023-10-24:
Similar however it'd be 0%=0%, 2% = x%, and 100% = y%. So off is off and any recognizable on would output a minimum pwm value, 100% could be set to max allowable pwm. 
Right now it hold pwm to 0 until you reach a pwm minimum requested set to 1.5. Now...ex 0=0 .5=0 1=0 1.5 =1.5. Id like 0=0, .01=1.5, .5=2 ,1 =2.5, 1.5 =3( this is just an easy low end ex). If max was adjusted thes numbers would be diff

phil-barrett, 2023-10-24:
Yes, from the Web builder would be very useful. Is it not enabled when I turn on Spindle sync? I have a tachometer but have been using VFD RPM readout.

terjeio, 2023-10-24:
> Similar however it'd be 0%=0%, 2% = x%, and 100% = y%

You want a lookup table or function for the converting percentages? Currently the [conversion is from RPM to PWM duty cycle](https://github.com/grblHAL/core/blob/04a5edcf8353f7294167d12d32e4b073d9be9ce4/spindle_control.c#L662-L740) and this code is called directly by the PWM spindle driver code and cannot be overridden by plugins. I may make them overridable if that helps.
It is possible to "fake" the RPM input as a percentage if you set $30 (max spindle speed) to 100, is this what you are doing? If so transforming the RPM input before it is sent to RPM -> PWM conversion might be what you are after.

> Is it not enabled when I turn on Spindle sync?

Ok, I'll add it.

> Is it not enabled when I turn on Spindle sync?

No, it is not. IIRC linearization was mainly aimed at (CO2) lasers?

phil-barrett, 2023-10-24:
> It is possible to "fake" the RPM input as a percentage if you set $30 (max spindle speed) to 100, is this what you are doing? If so transforming the RPM input before it is sent to RPM -> PWM conversion might be what you are after.

Probably better to set spindle max to 10000 and then you can express in 100ths %.  5005 is 50.05%

customsolutions, 2023-10-24:
That works on the max ouput end, however (i should specify), for laser if
it needs an minimum of 5% to fire once you do what stated above the minimum
now needs to be raised to 10%. So now in lightburn for ex instead of 0 to
100% you can only use 10% to 50 for etchingand maybe 80-100% for cutting.
What would be awesome is an absolute max for cutting that is set to produce
safe amperage for laser tube. Then preferably in light burn have a setting
that will be set on a layer to determine if its a cutting layer or an
engraving layer. Then 100% is set in lb and doesnt matter in grbl. However
tha absolute max is set in grbl. Next item would be a minimum need to
activate laser. 0= off and the next recognizable amount above zero starts
at laser on pwm/rpm/voltage for ex .01% = 5% then its mapped from there to
the afore mentioned absolute maximum.

customsolutions, 2023-10-24:
Here is an example. Lets say grbl is set to 10000rpm and lightburn also.... then 0%=0rpm, 5%=500rpm 100% = 10000rpm. Well 10k will prob burn up the laser so we set LB to a max of 5k, so max output from LB is what grbl sees as 50%(5k/10).
Now when we request 5% we only get 250rpm which is not emough to fire laser... so it would be nice to map .1% in grbl to the needed 500rpm. So the user sees 0=off .1% = laser on minimum and 100% = a safe maximum. Hope that helps.

terjeio, 2023-10-25:
I still do not get why setting $35 isn't working for the just above 0 RPM input, is your $31 setting > 0?

customsolutions, 2023-10-25:
Its been awhile since I played with it. Maybe tomorrow I can check the way
it reacts and confirm. Maybe Its a setting i had wrong. Is they a version o
should check also?

customsolutions, 2023-10-25:
Also is there a testing platform you are specifically using?

terjeio, 2023-10-25:
> Is they a version o should check also?

` $I` output contains build dates for both the core and the driver

> Also is there a testing platform you are specifically using?

The main one is STM32F466 on a Nucleo dev board, but I test on several others as well - depending on what has been changed.

terjeio, 2023-10-27:
FYI I have now changed the code for RPM > PWM conversion to make it easier to override the default functionality from plugins. However, these changes will not be made available from github until after I am back home in early december. Until then I can make the code available, on request, from the master subversion repo.

@phil-barrett I have added the option to enable spindle linearization from the Web Builder.

customsolutions, 2023-10-27:
Awesome, thanks for the update! Enjoy your trip.

phil-barrett, 2023-10-27:
> @phil-barrett I have added the option to enable spindle linearization from the Web Builder.

Thanks, have been building with it.  Hope you are someplace warm/nice.

customsolutions, 2023-11-22:
I decided to play with nucleo446 and a proteneerboard. When i use the web builder, and once connected to lightburn, I dont see $35, and when I try to set $32 to laser mode I get error 17. Under web builder general settings it does say spindle[pwm]. Am I 0 for 2 in webbuilder?

terjeio, 2023-11-22:
> I dont see $35, and when I try to set $32 to laser mode I get error 17.

My bad - default for the Protoneer board is spindle on/spindle dir/no PWM. I have changed this to spindle on/PWM for the Web Builder - so please try again.

customsolutions, 2023-11-23:
All the $$ menu options are there now. Not 100% that im getting Pwm....
does the webbuilder use the proteneer3xx.h board file and if so it looks
there is a spot to enable pwm( not sure your process on the backend). Also
looks like gpioa5 or d13 on arduino pinout? Thanks for your help. I hope
these things im finding are helpful too.

terjeio, 2023-11-23:
Is the PWM pin listed in the `$pins` output?

customsolutions, 2023-11-23:
Where is $pins located, I was looking in boardfile I assume is used for
proteneer shield.

phil-barrett, 2023-11-23:
In a GCode sender's MDI section (aka Console), enter $pins and you should get back the pin assignments used for that build.

customsolutions, 2023-11-23:
Wow, thats new to me. Thanks for that tip. I will give it a try and report
back.

customsolutions, 2023-11-23:
So when I check $pins in the console it shows PA5 = spindle pwm. However
upon testing it does not apear to be doing anything... PA6 is changing
state depending on m3/m5 commands.

terjeio, 2023-11-23:
The PWM output is on the spindle direction pin. Is this the pin you are checking?
There are several versions of the [Protoneer board](https://blog.protoneer.co.nz/arduino-cnc-shield/), including many clones. Some have an additional PWM pin assigned, seemigly swapped by the Z limit input. Which one do you have?

customsolutions, 2023-11-23:
I dont have the shield on, and was checking at the header directly...

phil-barrett, 2023-11-23:
The main way to tell the shields apart  - Grbl 0.9 and earlier do not have a labeled PWM spindle output pin, Spindle Enable is on D12 and Z limit is on D11.  Grbl 1.1 has PWM on D11, Z limit on D12 and they dropped spindle enable.  The protoneer shield 3.1 is a Grbl 0.9 board. For what it is worth, you can use the pins how ever you choose, there is no filtering or optos on that shield - the connectors go straight into the arduino headers.

customsolutions, 2023-11-24:
Im testing at the header on the nucleo64(446) board. Freshly built from
webuilder.

customsolutions, 2023-11-24:
Here is a question, on the webuilder should I be using (protoneer
v3/nucleo64 446) or (generic uno/nucleo64 446)?

terjeio, 2023-11-24:
You should be using the Protoneer version, if the Uno board has compatible pin mappings that could of course be used instead.

I have a F446 Nucleo board at hand and luckily PA5 is wired to an on-board LED. I flashed Web Builder build and it works for me in that the LED intensity varies with the commanded spindle speed.
Either your PA5 pin is damaged or you have laser mode enabled ($32=1)?

customsolutions, 2023-11-24:
That clears it up. I do have laser mode enabled. Because on previous boards
and versions for my laser pwm was used to control my laser intensity. Is
pwm moved? Or disabled in this version?

terjeio, 2023-11-24:
> Is pwm moved? Or disabled in this version?

For the Protoneer board: no and no.

When laser mode is enabled there will be no PWM output unless there is non-rapid motion ongoing. This is the same behaviour as legacy Grbl.

customsolutions, 2023-11-24:
Tried 2 diff nucleo boards with only a resitor and led wired as a tester. I
have verified that both pa5 and pa5 do turn on and off, however regardless
of polarity, or s setting pa5 is either off or so low i can balerely see
it. I have check the $30setting to make sure im sending the correct ramge
of values. Pa6 is alot brighter when it switches on. This is really
interesting. Both boards are brand new and freshly flashed today.

terjeio, 2023-11-24:
The on-board LED worked fine for me. Post your settings so I can check, use ioSender and select _Help > About > To clipboard_ and paste in a comment. Or check yourself with an oscilloscope.

customsolutions, 2023-11-24:
Thanks for all your suggestions. Currently not useing io sender. Im using
lightburn.  So could I post them here? Also my on board led also lights,
just very dim.

terjeio, 2023-11-24:
Just reset the settings with `$rst=*` and set `$14` as needed.

Here is how the LD2 lights up with `M3S1000` after I did that, IMO not "very dim":

[image]

customsolutions, 2023-11-24:
> Ill try that... thx.... ill post a pic too.

Ok, gave it a try. 2 diff pc's & 2 diff nucleo's same result. I did $rst
=*, then set $14=70 and $5=7, i had to disconnect then reconnect $x , then
m3s1000. I am curious what other settings you or may have set diff in web
build or our apps. Are you using iosender? Im going to try my revamped mg2
board(upgraded to stm32f103b and compiled grbhal in stmcubeide, for
comparison.

terjeio, 2023-11-24:
This is the firmware I tested with, built with default options in the Web Builder:

[firmware(3).zip](https://github.com/grblHAL/core/files/13462431/firmware.3.zip)

I use ioSender and putty for control and I only change $14 after resetting.

customsolutions, 2023-11-24:
Thanks, I'll check it out.

customsolutions, 2023-11-25:
Ok, so I had to do the same routine. Flash. Det $14=70, $5=7, LB related(do
a read, unplug and reconnect), $x. M3 s1000. But now much brighter. Not
sure whats diff.... any other settings in webbuilder diff you think?

terjeio, 2023-11-25:
> any other settings in webbuilder diff you think? 

Could be - which ones did you select?

customsolutions, 2023-11-25:
I believe I may have had probe, spindle linearization, and maybe reset as
estop.

terjeio, 2023-11-25:
Ok, spindle linearization is then the likely reason. Did you add setting values for that?

customsolutions, 2023-11-25:
I was under the impression those were the $35 etc.

terjeio, 2023-11-28:
> I was under the impression those were the $35 etc.

No, $66 - $69 are used for spindle linearization (if enabled). $35 and $36 is for min and max PWM output.

dJOS1475, 2024-01-18:
Hi All, I've just moved over from FluidNC (didnt play nice with my prefered sender OB Control), and Im finding the PWM spindle control to not work very well. In FluidNC I was able to put in a simple speed map and this worked perfectly. I cant use the SPINDLE_LINEARIZATION module as my GRBL_SPD controller doesnt support the features AFAIK.

Here is my current grblHAL config vs FluidNC - am I missing something obvious?

`grblHAL:

$30=30000.000 ;Maximum spindle speed, RPM
$31=10000.00 ;Minimum spindle speed, RPM
$32=0 ;Laser-mode enable, boolean
$33=1000.0 ;Spindle PWM frequency
$34=0.0 ;Spindle off Value
$35=1.0 ;Spindle min value
$36=100.0 ;Spindle max value

FluidNC:

PWM:
  pwm_hz: 1000
  direction_pin: NO_PIN
  output_pin: gpio.32
  enable_pin: gpio.27
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 10000=1.000% 30000=100.000%`

cheers Derek

terjeio, 2024-01-19:
@dJOS1475 
> I cant use the SPINDLE_LINEARIZATION module as my GRBL_SPD controller doesnt support the features AFAIK.

Did you try it?  The GRBL_SPD controller takes a PWM input so it should work?

dJOS1475, 2024-01-19:
> @dJOS1475
> 
> > I cant use the SPINDLE_LINEARIZATION module as my GRBL_SPD controller doesnt support the features AFAIK.
> 
> Did you try it? The GRBL_SPD controller takes a PWM input so it should work?

No, I got around the issue I was having by setting the minimum to 2. I’m not entirely sure why that worked tho (Jon the designer suggested it).

dJOS1475, 2024-01-28:
> @dJOS1475
> 
> > I cant use the SPINDLE_LINEARIZATION module as my GRBL_SPD controller doesnt support the features AFAIK.
> 
> Did you try it? The GRBL_SPD controller takes a PWM input so it should work?

OK, I thought I had it solved, but I didnt. So I recompiled grblHAL with the Spindle RPM linearization module and ran your software for it. Frankly Im confused!!I know im an idiot who cant code to save himself, but I do not understand how your system is supposed to work??

Please just implement a simple speed map, like FluidNC, and you solve a lot of ppl a lot of trouble eg:

Speed_map: 0=0.000% 10000=1.000% 30000=100.000%

Honestly, if you assume ppl are as dumb as I am, you'll make life easier for everyone.

terjeio, 2024-01-28:
Looking at this again I believe the $30-$36 settings above should be equivalent to the FluidNC speed map. So what is not working?

dJOS1475, 2024-01-28:
> Looking at this again I believe the $30-$36 settings above should be equivalent to the FluidNC speed map. So what is not working?

Tbh, I have no idea what happened. I recompiled the firmware and zeroed out the linear settings and it started behaving. 

Bizarre!

closed 2024-04-16 by terjeio