grblHAL/core #385: Keyboard Jogging does not work when using Ethernet-to-uart module

fireup924, 2023-10-25:
When using IOSender and communicating via Ethernet-Uart module [,](https://www.waveshare.com/uart-to-eth-b.htm), the Keyboard jog does not work and gives a "G-code words consist of a letter and a value. Letter was not found." error.

I traced it down to this section of protocol.c on line 890 `hal.stream.cancel_read_buffer();` commenting this line out and adding
`if (state_get() & STATE_JOG) // Block all other states from invoking motion cancel.
                system_set_exec_state_flag(EXEC_MOTION_CANCEL);`

I'm not sure if this is the right fix for this? I seems the keyboard jog command for IOSender is sending a 0x85 and a [$J=](url) in the same line and cause the hal.stream.cancel_read_buffer() to trim some of the Jog string...

terjeio, 2023-10-26:
I assume this is with the RP2040 driver. The issue seems to be that the code for flushing the RX buffer that also attempts to empty the RX FIFO upsets something. Commenting out the [FIFO flushing part](https://github.com/grblHAL/RP2040/blob/9c3a039a0b84aa454b0cdcfd13093f88178ef4cc/serial.c#L216-L217) resolves the issue.
I guess  I'll have to figure out how to empty the FIFO without causing side-effects.

fireup924, 2023-10-26:
Yes, it's on a RP2040. I wasn't sure if it's was only RP2040 so I tested it on ESP32 and Teensy4 and they don't have the same issue.