grblHAL/core #389: Where to find grblHAL pinout?

AlexeyHaritonov, 2023-11-04:
Hello all,

I'm try to compile 
grblHAL driver for STM32F7xx
I have some nucleo-144 board to test it
But no info on the pinout....
Any ideas?

I think it can be very useful to have a list of pinouts for every platform. 

Best Regards,
Alexey

terjeio, 2023-11-04:
Pinouts are in `*_map.h` files, if compiling with the [Web Builder](http://svn.io-engineering.com:8080/) click the _Board map_ button to open the map for the selected board.
FYI the STM32 drivers map files are located in the _Inc_ folder.

AlexeyHaritonov, 2023-11-04:
Thanks
What about ganged motors? Can't find it in Board map

terjeio, 2023-11-04:
Ganged/autosquared motors are defined in _my_machine.h_, allocated automagically from the motor definitions in the board map (M3_*, M4_*, ... definitions). Use the `$pins` command after flashing to list the bindings.

closed 2023-11-04 by AlexeyHaritonov