grblHAL/core #395: Homing of rotary axis without hard limit functionality

karoria, 2023-11-23:
Hi @terjeio 
I want to configure homing for rotary axis A, for which I have made necessary changes to $44 and $45 settings ($44=4, $45=11) which works as expected but I can't figure out how to disable A axis limit switch being detected as Hard Limit while in operation. I tried to go through wiki and old issues but could not get the answer. Please help.
Thanks.
Ravi

terjeio, 2023-11-23:
$21=5 disables hard limits for rotary axes (as enabled by $376).

karoria, 2023-11-25:
Thanks. Since I was using old version of grblHAL, I am building new version from scratch. I am able to successfully compile and flash but no any communication works (ethernet or usb). Any clue where I am going wrong? My board is having custom board map and my primary communication means is ethernet. Is there any manual patching process involved now like it was earlier for lwip and SD libraries? Can it affect USB communication too?

terjeio, 2023-11-25:
> Any clue where I am going wrong?

Changes to (serial) port handling? Check the changes done to one of the board maps?
Old and new versions mixed up?

> Is there any manual patching process involved now like it was earlier for lwip and SD libraries

No, not any longer - patched versions are now provided (as submodules).

Can you try a Web Builder build?

karoria, 2023-11-25:
In board maps, I could not find any obvious things. Then I tried Web Builder and chose T41BB5X Pro as sample board and observed that the builds with Macros Enabled (any of the 3 listed options) doesn't communicate with USB cable. I haven't checked with Ethernet though as I am unaware of the IP settings of that particular board.

terjeio, 2023-11-25:
> builds with Macros Enabled (any of the 3 listed options) doesn't communicate with USB cable.

Do you get a response when sending a real time report request: `?`

> I also got all communications working if I disable macros in my_machine.h and compile with PIO. Is that setting disabled will also stop ATC routine via expressions?

No, there are two different macro implementations. The one you use for ATC is enabled when SD card and expressions are enabled.

karoria, 2023-11-25:
>Do you get a response when sending a real time report request: ?

No. It doesn't get connected at all! Disabling macro has sorted out my problem as of now as I don't use input or keypad based macro execution. With Web Builder you will be able to reproduce the problem as above.

terjeio, 2023-11-25:
> No. It doesn't get connected at all!

So the USB port does not show up?

> With Web Builder you will be able to reproduce the problem as above.

Ok, I'll add this to my todo list when I am back home.

karoria, 2023-11-25:
>So the USB port does not show up?

The USB port shows up on my MacBook. I use gSender Edge version to connect to my grblHAL board as I don't have Windows machine. After connecting it immediately disconnects back. Once I disabled macro and flashed, it works as expected on the same software. Now it also communicates via Ethernet and works as expected. Ethernet magjack was also not showing up lights when flashed with macros enabled. Now it works as expected.

karoria, 2023-11-27:
BTW $21=5 disables hard limits for rotary axes as you have mentioned. Thanks.

terjeio, 2023-12-11:
> builds with Macros Enabled (any of the 3 listed options) doesn't communicate with USB cable.

This is a driver bug, fixed in the next commit.

terjeio, 2023-12-15:
The driver bug is fixed in the latest commit.

closed 2023-12-15 by terjeio