grblHAL/core #397: How to hide littlefs from showing up in FTP  as well as files list

karoria, 2023-11-25:
Hi @terjeio 
I liked the old behaviour of not showing up webui files in littlefs in either FTP/WebDAV or files list populated by "http://URL/upload?path=/" I want the users of machine don't delete the webUI files by mistake. I would like to use only "http://URL/?forcefallback=yes" for webUI file management in flash memory. This behaviour was there in old builds but updating to new build has made those files visible in all places.

terjeio, 2023-11-28:
A new setting, $650, for filing system options will get a flag for "hiding" the littlefs mount when listing the contents of the root directory in the next core commit.

karoria, 2023-11-28:
Thanks @terjeio 
I will be eagerly waiting for this new commit.

terjeio, 2023-12-15:
It is [available now](https://github.com/grblHAL/core/blob/master/changelog.md#20231210).

closed 2023-12-15 by terjeio

karoria, 2023-12-16:
Thanks @terjeio ...