grblHAL/core #413: Debug statements inside macros prints immediately when read

karoria, 2023-12-22:
Hi @terjeio 
I noticed that (debug,________) statements inside any macro file prints immediately when they are read by planner and not when they are actually executed. I mean they are out of sync and hence doesn't solve the purpose of debugging. Am I missing something or there is work in progress in this matter?
Regards,
Ravi

terjeio, 2023-12-23:
For now add a short G4 delay before the debug message.
I'll consider adding it to the data sent downstream to the planner/step execution later but it might be a bit complicated since, for motion commands, there can be many line segments that has to complete before the message can be output.

karoria, 2023-12-25:
Ok. Does a G4 delay make the debug statements synchronous with previuos and next commands? In which cases adding G4 delay might help?

terjeio, 2023-12-25:
Yes, try `G4P0.1` for a 100 ms delay.

closed 2026-02-27 by terjeio