grblHAL/core #418: Perhaps a GRBL Hal Issue - Perhaps not.

Klippert, 2024-01-01:
Hello, 

First and foremost, thanks for a formidable piece of software!

I have a problem, that my holes does not get round. If I bore a 6mm hole it becomes a 45 degree oval measuring about  6,15mm and 5,69mm - and I've checked that my acceleration settings are all good.

The twist is if i mill up a rectangle i get about 0,05 mm accuracy repeatable. 

I run with closed loop steppers, so if a steps were lost, they stop and go in error mode.

I've checked my GRBL hal config, and the settings for acceleration on X and Y are ok.

I'd love to hear any suggestions on what could be wrong... Any suggestions are most welcome. 

best regards, happy newyear and merry christmas :-)
Kristoffer

andrewmarles, 2024-01-02:
So far, around 95% of these kinds of issues I've seen with users are due to incorrect settings on the step pulse delay.  Check the manual for your drivers and ensure that you are meeting the setup and hold requirement for the step/pulse transition.

Klippert, 2024-01-02:
> So far, around 95% of these kinds of issues I've seen with users are due to incorrect settings on the step pulse delay. Check the manual for your drivers and ensure that you are meeting the setup and hold requirement for the step/pulse transition.

Interesting! - So if the driver don't get the impulse correctly, the motor won't make the correct number of steps? 
But then it would generally lose position, and everything else based on that homing should be off, right? And as far as I can measure, it's not? Or am i missing something?

karoria, 2024-01-02:
@Klippert When an axis direction changes, this delay comes into picture. Your motor drivers need some time when that axis changes direction that should be specified in their datasheets which you can feed in step idle delay. That may solve your issue. In your case, hole is the feature when motors need to change direction frequently and the wrong setting of this parameter may lead to missed steps or some steps counted in previous direction due to driver's hardware latency. This can make oval holes. I assume your machine is mechanically squared in good accuracy.

hanke-cnc, 2024-01-02:
Certainly not a problem with GRBL, which is mature and reliable. Recently, a customer reported to me that when milling a square, the length would decrease by 4mm. However, manual measurements during operation were correct. After their investigation, it was determined that the issue was caused by a loose coupling in the drive system.

Klippert, 2024-01-02:
> @Klippert When an axis direction changes, this delay comes into picture. Your motor drivers need some time when that axis changes direction that should be specified in their datasheets which you can [...]

Thanks - So Step idle delay - is that a GRBL Hal setting that essentially tells the machine to wait  for the motor to come to a stop before changing direction? 
I looked in the Grbl hal config, and the only one i found was essentially turning off the steppers after they come to a halt. I suppose that's not what we're looking for here.

I did go through all my drivers, and re tuned them with new settings for currents, position follow limit which i lowered to the equivalent of 0,1mm. it was a whopping 0,5mm. So now my drivers should fail, if the steppers can't keep up with my feedrate.

Also i tried boring some holes at 400mm/min and at 1000mm/min and not entirely unexpectedly the slower holes were significantly rounder than the faster ones. Also the smaller holes were more oval than the larger ones. Which leads me to really ponder, if it's not the acceleration of the motors that is not working as intended.
In Grbl config both the X and Y axis has the exact same settings aside from the max speed, where my Y (gantry) can move quite a bit faster than my X-axis, as it's powered by two motors.

Here is a picture of the holes: The leftmost pattern was done at 400mm and the rightmost at 1000mm. Especially the smaller holes get quite far out of round...

[image]
[image]

Again, thanks for all the answers so far, and any further help would be greatly appreaciated.

Best
Kristoffer

andrewmarles, 2024-01-03:
[image]

karoria, 2024-01-03:
Try 5 microsecond pulse delay and do the same job. Leadshine DM542 datasheet tells that (see remark 'b'). Yours might be different. Why not set it by trial and error if you don't have that parameter for your hardware. Going faster make more ovality can also be logically understandable if you have this issue.[image]

Klippert, 2024-01-04:
I reajusted the controllers to the following settings. Now i have a problem that when i reach an endstop, the decelleration is too abrupt. is there a way of letting it decelerate over say 1 or 2 mm? I use optical endstops, so I can afford overtravel. 
Then I'll make the experiment with the pulse delay as you suggest. it souds like it could be the issue.

Anyway, this is the config as well as the motor test result. as you can see it's a bit trailing when moving at full speed. - i know very little about it so if anyone have experience, i'm  all ears ;)

[image]
[image]

Klippert, 2024-01-06:
Hello Again,
I just did a test where I lowered the acceleration on the X and Y Axis to 25mm/sec2 - from 250mm/sec2. This produced holes with much tighter roundness - about 0,04mm out on a 6mm hole and 0,06 out on a 15mm hole. which is approaching something acceptable :) - at 400mm/sec. Now i'm going to do some further tests with low acceleration setting, and see how it works for different feeds. - well knowing that for small holes, the feed will never get very high due to the low acceleration. 

I've also added the 5us pulse delay, however I still need to do the test with the old accelleration setting to see if that does anything.

So, i'll be back with more experiement results.

Also, when homing, the motor stops when it hit the endstop - is there a way of setting an accelleration (deceleration) for this stop? I'm using optic endstops, so I can afford some overtravel.

Thanks again for all your kind help.

terjeio, 2024-01-06:
> This produced holes with much tighter roundness

IMO this is a sign that the machine is not stiff enough for the higher acceleration/feed rates.

> at 400mm/sec

That is an insane feed rate...

> the feed will never get very high due to the low acceleration

Try increasing the size of the planner buffer ($398) - it may help. FYI the default was increased from 35 to 100 recently.

> Also, when homing, the motor stops when it hit the endstop - is there a way of setting an accelleration (deceleration) for this stop? 

Currently not, step output is abrubtly cut off when a limit switch is hit.

Klippert, 2024-01-06:
Hello,

First a correction. 400mm/sec was meant to be 400mm/min 😂

Thanks for the suggestion on the planner buffer. I’ll try increasing that.

Would the deceleration on homing be an option to add? I guess even a
decceleration over half a mm would go a long way. Also it would be more
gentle on the machine. Currently my controller says that it can’t keep the
motor in the limits of how far it can get out of alignment, and that means
it over travels with more than 0,05mm. How much more I can’t tell.

Thanks again for all your help.

karoria, 2024-01-06:
I also feel the need of "decelerate while homing" if it is possible.

terjeio, 2024-01-07:
Decelerate while homing will be a major change, especially if homing more than one axis axis at a the same time (or an auto squared axis) since only one motor has to be decelarated while the other(s) is continuing moving. And depending on the limit input device motion has to be limited in order not to destroy it since not all allow for too much overtravel?

Klippert, 2024-01-07:
Yes. That’s good points. I guess one could set a number of steps to
decelerate over after limit switch triggering. If one is using a mechanical
limit switch it’s a risky feature to implement.

I just ended up lowering homing speed.

Best
Kristoffer

closed 2024-04-16 by terjeio