grblHAL/core #421: Can’t write firmware to MKS Tinybee v1.0

Davinjw, 2024-01-07:
Can’t seem to get firmware installed. Getting a “no xmc flash detected”error

any ideas?
Thanks in advance!

[image]

Davinjw, 2024-01-08:
Disregard/delete. I didn’t check the boxes on the left -facepalm-

grzesiekn, 2024-02-03:
Hi, I can flash the board using files from web builder, but after connecting via putty I get gibberish. I am using 115200 8N1.

terjeio, 2024-02-03:
@grzesiekn This works for me (with a dev board):

[firmware.zip](https://github.com/grblHAL/core/files/14151923/firmware.zip)

[image]

grzesiekn, 2024-02-03:
It works, thank you.

closed 2024-04-16 by terjeio