grblHAL/core #430: If you set MODBUS you lose the IOSender connection

MrKapim, 2024-01-20:
Hi Terje, I'm trying to connect the spindle in RS485 (MODBUS), but I'm not getting the connection to IOSender. If I set in MYMACHINE the options of SPINDLE0_ENABLE=SPINDLE H100 (this is my Chinese model) and MODBUS_ENABLE=1, the connection to IOSENDER is not established. IOSender opens and reports that there was no response from the controller and closes. What am I doing wrong?

terjeio, 2024-01-21:
This is likely an issue with the board map and could be a regression. So which driver/board?

MrKapim, 2024-01-21:
I'm using the STM32H743 on BTT SKR3 EZ

MrKapim, 2024-01-21:
I've made changes to MYMACHINE.H and now I have IOSender communicating but I get message 14 every time I send command to the axis (M3 or M4). I don't know if this change is correct (setting the serial port to MODBUS and VFD_ENABLE=6). I put VFD_ENABLE as 6 only because the H100 is set to 6...
[image]

MrKapim, 2024-01-21:
An observation...  I turn the set off and on again. On my RS485 interface, the RX led pulses 2 times and is off. When I open and connect the IOSender, the RX led keeps pulsing (about 500ms), which shows that the connection is being "queried". Is this correct? Shouldn't it be "pulsing" from the start (when turning on)?

MrKapim, 2024-01-21:
I am sorry. Is the TX led that's pulsing not RX

MrKapim, 2024-01-21:
Here my pins...    
[image]

MrKapim, 2024-01-21:
Some more information to help you think...    I've tested the 3 serial ports set and if I change to any of the others (1 or 3) the IOSender doesn't communicate. It (IOSender) only communicates if I use port 2 (SERIAL1_PORT)

terjeio, 2024-01-21:
> Is the TX led that's pulsing not RX

This means that the VFD is not responding, is it set up correctly?

terjeio, 2024-01-21:
And wiring could be wrong, typically RX/TX should be crossed (RX -> TX, TX -> RX) for serial.

MrKapim, 2024-01-21:
I set up the drive according to the manual...  F163 = 1 (address)   F164 = 2 (19200 baudrate)    F165 = 3 (8N1 for RTU)

MrKapim, 2024-01-21:
> And wiring could be wrong, typically RX/TX should be crossed (RX -> TX, TX -> RX) for serial.

I had already reversed the pins to test and no results

terjeio, 2024-01-21:
[F001/F002](https://forum.onefinitycnc.com/t/zhong-hua-jiang-cnc-spindle-110v-1-5kw-and-h100-vfd-need-modbus-settings/17319/8) has to be changed?

MrKapim, 2024-01-21:
Yes, they have been changed to F001 = 2 (communication port) and F002 = 2 (communication port)

terjeio, 2024-01-21:
Is it a 100% compatible H100 VFD? And has it the RS-485 interface installed?
IIRC a Modbus compatible interface should respond with an error if it does not understand a message - since the RX LED is not blinking no response is sent. This _could_ be due to baud rate mismatch. There is a lot more discussed in the thread I linked above, I do not know if any of that is relevant to your issue.

FYI the code was recently fixed and verified - see [this comment](https://github.com/grblHAL/Plugins_spindle/issues/22) so I assume it is ok. And when I test with my simulator everything seems to be ok, this with a STM32F4xx driver.

MrKapim, 2024-01-21:
Hi Terje. It's working! I replaced the RS485 interface. Thank you for your help, patience and great work...

MrKapim, 2024-01-23:
Hi Terje, I encountered two errors...  
Error 1 - When sending M3S5000 for example, the spindle starts and returns the information that it is at 5000 rpm. Then I send M4S5000, I don't get any feedback, the spindle keeps spinning in the same direction. But if I do it the other way around (send M4s5000 and wait for the spindle to rotate) and then send M3s5000, the spindle responds correctly by reversing the direction of turn. 
Error 2 -  When sending M5, the spindle just "turns off" and it stops by inertia. But I adjusted the parameters and I have a brake resistor installed and if I do this same operation through the panel, the spindle responds with a quick stop. Can you help me with these little problems? Thanks

MrKapim, 2024-01-23:
Oh yes, another detail...   IOsender updates the rotation of the spindle on the screen, but when I send m3 or m4 with S0, or send m5, the spindle stops but it keeps the value of the rotation it was...   
[image]
[image]

terjeio, 2024-01-23:
Error 1: this is how the VFD reacts to a direction change since the command sent is correct. So perhaps a VFD bug, or maybe it is controlled by a F setting? Anyway it would be best to stop it before changing direction?
FYI Grbl/grblHAL has no interlock preventing direction changes when the spindle is running.

Error 2: Is this controlled by F026? And F030? Other settings? Better to ask your supplier since I do not know and have to guess from what I read in the manual. And I do not have a VFD to try out settings with.

> IOsender updates the rotation of the spindle on the screen, but when I send m3 or m4 with S0, or send m5, the spindle stops but it keeps the value of the rotation it was...

This is per design - I display the last entered RPM when stopped to avoid entering a new RPM every time starting the spindle manually. Or does it display the last RPM received from the VFD when stopped? It should not.

MrKapim, 2024-01-24:
Huuummm...   Error 1: I did new tests and explaining better...  It does not make the change of direction from M3 to M4 only if "s" is the same value, and from M4 to M3 it makes the change regardless of the value of "s". 
Error 2: Yes. F026 = 0, F028 = 0.2, F030 = 0.2 and the times for deceleration...  but I'm sorry! I went to check the values and they were high. The stop is OK
And about the values from RPM on IOsender screen...  It shows/updates the rotation value by decreasing and then updates to the last sent value
And about your comment about stopping and then reversing the rotation... I agree that it's the best way to use it, I'm just demonstrating what I've found. For the application I will be using, I am "forced" to stop and then reverse the direction of rotation of the spindle...

MrKapim, 2024-01-24:
Thanks again!!!

closed 2024-04-16 by terjeio