grblHAL/core #435: Not an Issue: How to limit jog area irrespective of machine travel?

karoria, 2024-01-28:
Hi @terjeio 
Is there a way so that we can restrict jog area even though my machine travel is larger? This is because I don't want to jog the machine in my ATC tool rack area. When tool change macro runs, then only machine should go to that area. This will reduce chances of accidents with tool rack while jogging. I have noticed $40 setting but I am not sure if that can solve this issue as it also allows machine to jog as per values in axis travel parameters.

terjeio, 2024-01-28:
$40 enables clipping to joggable envelope, by default this calls [apply_jog_limits()](https://github.com/grblHAL/core/blob/4a7090d70aadddcd426d5326bb252fe4dc3ee40c/machine_limits.c#L754-L782) via the [grbl.apply_jog_limits](https://github.com/grblHAL/core/blob/4a7090d70aadddcd426d5326bb252fe4dc3ee40c/machine_limits.c#L789) function pointer. This means that `apply_jog_limits()` can be overridden by plugin code by redirecting the function pointer.

karoria, 2024-01-28:
Ok. Thanks👍

closed 2024-01-28 by karoria