grblHAL/core #436: MKS Tinybee and Pendant??

lons81, 2024-01-28:
Hi,
The MKS Tinybee with GRBLhal is absolute fantastic...
Now i try to use my ESP32 Pendant, but i didnt get it to work.

I think i need the Keypad plugin, but in Web Builder its not checkable.
So how can i tell grblhal to use Uart2 for my pendant?

With FluidNc it worked great.

Thanks, Lons

terjeio, 2024-01-29:
> Now i try to use my ESP32 Pendant, but i didnt get it to work.

Do you have a link to it? Or did you make it yourself?

> So how can i tell grblhal to use Uart2 for my pendant?

First I have to make the port available, it is on my todo list for the ESP32 driver.

terjeio, 2024-01-29:
The [keypad plugin](https://github.com/grblHAL/Plugin_keypad) can now be enabled in the Web Builder. I am not sure that solves the issue for you since grblHAL only allows input from one source at a time unlike FluidNC that AFAIKT allows input from multiple sources. FYI the keypad plugin uses the 0x8B character to take full control of the controller (by switching the input stream).

closed 2024-04-16 by terjeio