grblHAL/core #445: Web builder loss of EEPROM settings on json load

calabr, 2024-02-15:
Playing with firmware for STM32F4xx I found the Web builder always resets EEPROM plugin to "no" after loading .json file.
Any STM32 with flash enabled works the same.
If firmware config made in web builder, it's creating correct firmware, but if .json saved locally and than loaded EEPROM always set to "no". 
  Test to reproduce:
  STM32F4xx
     Devtronic CNC
go to plugins - EEPROM 24x16 enabled by default
save .json
load .json
  EEPROM become NO

   The same behaviour also with STM32F1xx

terjeio, 2024-02-16:
Thanks for reporting, should be ok now.

calabr, 2024-02-16:
Thank you for fast reaction!
 Now it's works correctly.

  Web builder page should be forced reloaded ( with SHIFR reload) to renew HTML page

closed 2024-02-16 by calabr