grblHAL/core #446: Rotational Axis behaving as linear

Banchad2, 2024-02-15:
I have a 5 axis compile going on a flexihal where I am intending to use the B axis as a ATC carousel. I'm seeing behaviour where the B axis is not behaving as a rotational axis instead as a linear. I CAN bodge it to work but I am not keen that going forward as I have a feeling that will cause problems down the road.

I've attached a screenshot below of the settings along with a backup of the settings file(hopefully I'm just being an idiot and have settings incorrectly configured)

[image]

[image]

[settings.txt](https://github.com/grblHAL/core/files/14303346/settings.txt)

terjeio, 2024-02-16:
What is not behaving as you would expect?

Banchad2, 2024-02-16:
> What is not behaving as you would expect?

As I understand it the rotary axis should operate in degrees.  IE Servo is configured to 8000pul/rev, 10:1 gearbox means that actually translates to 80,000. Setting grblhal step/rev to 80,000 and jogging by 1 unit(expected to be degree) makes a full 360 degree rotation. Settings above are the settings to make 1 unit jog = 1 degree rotation but this results in the rotation speed being wildly out from reality. Its also able to rotate past 360 and does not wrap back to 0.

terjeio, 2024-02-16:
The axis units shown are wrong due to the build including all spindles. There is a new spindle type among them, _Stepper_, that changes the units, and does so even when not enabled. So read "deg" instead of "rev" and configure accordingly. I'll fix this in the next commit.

> Its also able to rotate past 360 and does not wrap back to 0.

This is how it should be, a rotary should be able to move more than one rotation and still keep track of the tooltip position in 3D space.

closed 2024-04-16 by terjeio