grblHAL/core #448: CMD_STOP doesn't produce any response

nickshl, 2024-02-20:
Description for CMD_STOP says:
> 0x19 (stop) is similar to the legacy 0x18 (reset) but does not perform a warm reset of the controller leaving more internal states intact.

Which sounds useful. However, controller doesn't react in response other that state change from "Run" to "Idle". Any other kind of stop produces alarm(_physical button makes Alarm 10 in my configuration, CMD_RESET produces Alarm 3_). Also when we feed gcode to the grblHAL, it accept some number of commands responding "ok" each time until it can't accept commands anymore, after which grblHAL will hold "ok" response until it ready. If command CMD_STOP is sent, grblHAL will never respond "ok" to any queued commands(_which make sense since those commands is purged_).
As result there more guess work for sender: if CMD_STOP was send and state changed from "Run" to "Idle", sender should **assume** that CMD_STOP command is executed and any queued commands was purged. Is it possible to make some explicit response to CMD_STOP command?

nickshl, 2024-02-22:
@terjeio , by the way... I want to send you something. But I need your address. I found one at your website(_in oldstuff section_), but I don't think it is complete. GitHub doesn't allow private messages and sharing address publicly is not an option. How can I contact you?

terjeio, 2024-02-22:
> Actually senders already broken.

By broken I mean crashing or misbehaving in unpredictable ways. I will add handling of the stop message to ioSender - but give me a bit of time...

> How can I contact you?

Look at the [old stuff](http://www.io-engineering.com/oldstuff/index.htm) again.

nickshl, 2024-02-23:
> Look at the [old stuff](http://www.io-engineering.com/oldstuff/index.htm) again.

Sounds about right. I found this address before somewhere associated with Io Engineering.
This is your package:
[image]
You can track it on [USPS website](https://tools.usps.com/go/TrackConfirmAction?&tLabels=UA106950029US) and probably on Norway post website too at some point. 

If I understand correctly, you shouldn't pay any taxes or fees because we both private individuals and value is less than 1000 NOK(_I declared $60 which probably is about right for components cost_).

Have fun! 🙂
You can find my phone number on the package - text me in case of any questions.

terjeio, 2024-02-28:
The due for VAT limit in Norway has been changed to 0... But sometimes they let it through without charges.

Anyway, I just uploaded [new edge versions](https://www.io-engineering.com/downloads/) of ioSender that should handle the `Stop` message.

nickshl, 2024-02-28:
> The due for VAT limit in Norway has been changed to 0... But sometimes they let it through without charges.

On [this Posten page](https://www.posten.no/en/customs/receiving-from-abroad) there a link to Norwegian customs "[More about gifts sent from abroad](https://www.toll.no/en/goods/gifts/)" and it says:
> You may receive a gift shipment **free from duties and taxes** provided the value is NOK 1 000 or less and provided the shipment is sent from a private individual residing abroad to you as a private individual in Norway.

Hopefully they will charge nothing to you 🙂

nickshl, 2024-03-06:
@terjeio why do you have [such small mailbox](https://sporing.posten.no/sporing/UA106950029US)? 😄

nickshl, 2024-03-08:
Hi @terjeio , according tracking your [package is waiting for pick up](https://sporing.posten.no/sporing/UA106950029US). By the way, I found "bug" on Posten website: it uses browser local time(_not a Norwegian time_) for busy graph.

terjeio, 2024-03-08:
Ok, I'll pick it up tomorrow when I go shopping.

terjeio, 2024-03-11:
Thanks, a nice pendant you have made. I'll try it out with my mill.