grblHAL/core #449: Homing individual rotary axis

paukstelis, 2024-02-22:
I have a machine with two rotary axes (A and  B) and I am trying to figure out how to home a single rotary axis. I have DEFAULT_HOMING_SINGLE_AXIS_COMMANDS set to On, but if I give a $HB I get error 5. Further, I thought I would just set the homing cycles so the first homing cycle  is 16, and that indeed starts homing the B axis, but even with all other  homing cycles set to 0, homing fails (B axis does stop moving when hall sensor trigger fires). Not sure what I am missing here?

nickshl, 2024-02-22:
Have you tried something like this:
`G28 G91 B0`

paukstelis, 2024-02-22:
> Have you tried something like this: `G28 G91 B0`

That doesn't activate homing, but if I trigger the sensor after issuing the command there is a "hard limit" error.

Changing $22=3 (or some other values) let's me initiate homing with $HB. It also gives a `HOME` message when that axis is triggered, but the machine seems to be trying to home something else after it is triggered, as it stops responding to `?`. Eventually it will respond with homing failed.

```
Recv: ok
Send: $HB
Recv: <Home|WPos:0.000,0.000,0.000,0.000,0.000>
Send: ?
Send: ?
Send: ?
Send: ?
Send: ?
Send: ?
Send: ?
Send: ?
Send: ?
```

paukstelis, 2024-02-22:
I think you intended for those $23 to be $22, correct?  I have to set $22=3 in order for homing to be enabled and for it to start individual axis homing, which I believe is correct.

I do have homing enabled and everything is correct with the pin and wiring. I can see the Pn triggered in status report. With $22=3, I can do a $HB and it will begin homing. When the sensor is triggered the axis stops moving, however, it just continues doing some kind of homing. No other axis moves, but it will eventually timeout. It is as if triggering the pin is not ending the homing after the axis is triggered. I have pull-off set to 0, as this is hall sensor and not limit switch.

Here are the rest of the settings:

```
Recv: $0=10.0
Recv: $1=255
Recv: $2=0
Recv: $3=21
Recv: $4=31
Recv: $5=26
Recv: $6=1
Recv: $9=1
Recv: $10=19
Recv: $11=0.010
Recv: $12=0.002
Recv: $13=0
Recv: $14=7
Recv: $15=0
Recv: $16=0
Recv: $17=0
Recv: $18=0
Recv: $19=0
Recv: $20=0
Recv: $21=5
Recv: $22=3
Recv: $23=1
Recv: $24=500.0
Recv: $25=500.0
Recv: $26=100
Recv: $27=0.000
Recv: $28=0.100
Recv: $29=0.0
Recv: $30=1000.000
Recv: $31=0.000
Recv: $32=1
Recv: $33=1000.0
Recv: $34=0.0
Recv: $35=0.0
Recv: $36=100.0
Recv: $37=0
Recv: $39=1
Recv: $40=0
Recv: $43=1
Recv: $44=16
Recv: $45=1
Recv: $46=0
Recv: $47=0
Recv: $48=0
Recv: $62=0
Recv: $63=3
Recv: $64=0
Recv: $65=1
Recv: $100=400.00000
Recv: $101=250.00000
Recv: $102=400.00000
Recv: $103=35.55000
Recv: $104=35.55000
Recv: $110=1000.000
Recv: $111=1000.000
Recv: $112=1000.000
Recv: $113=1000.000
Recv: $114=1000.000
Recv: $120=100.000
Recv: $121=100.000
Recv: $122=100.000
Recv: $123=100.000
Recv: $124=100.000
Recv: $130=250.000
Recv: $131=200.000
Recv: $132=600.000
Recv: $133=360.000
Recv: $134=360.000
Recv: $140=850
Recv: $141=500
Recv: $142=850
Recv: $143=1000
Recv: $144=850
Recv: $150=16
Recv: $151=16
Recv: $152=16
Recv: $153=8
Recv: $154=8
Recv: $180=25.0
Recv: $181=25.0
Recv: $182=25.0
Recv: $183=25.0
Recv: $184=25.0
Recv: $190=500.0
Recv: $191=500.0
Recv: $192=500.0
Recv: $193=500.0
Recv: $194=500.0
Recv: $200=120.0
Recv: $201=22.0
Recv: $202=10.0
Recv: $203=22.0
Recv: $204=22.0
Recv: $210=50
Recv: $211=50
Recv: $212=50
Recv: $213=50
Recv: $214=50
Recv: $220=22.0
Recv: $221=20.0
Recv: $222=22.0
Recv: $223=22.0
Recv: $224=22.0
Recv: $338=29
Recv: $339=5
Recv: $341=0
Recv: $342=30.0
Recv: $343=25.0
Recv: $344=200.0
Recv: $345=200.0
Recv: $346=1
Recv: $372=0
Recv: $376=3
Recv: $384=0
Recv: $398=35
Recv: $481=0
Recv: $486=0
```

Build info:
```
Recv: [VER:1.1f.20240204:]
Recv: [OPT:VNMHASLW,35,1024,5,0]
Recv: [AXS:5:XYZAB]
Recv: [NEWOPT:ENUMS,RT+,HOME,SED,TMC=29]
Recv: [FIRMWARE:grblHAL]
Recv: [SIGNALS:RHS]
Recv: [NVS STORAGE:*FLASH]
Recv: [DRIVER:LCP1769]
Recv: [DRIVER VERSION:2400205]
Recv: [BOARD:BTT SKR V1.4 Turbo]
Recv: [AUX IO:0,3,0,0]
Recv: [CLUSTER:16]
Recv: [PLUGIN:LightBurn clusters v0.06]
Recv: [PLUGIN:Trinamic v0.15]
```

terjeio, 2024-02-22:
>  I have pull-off set to 0

Increase it, a pull off is needed disengage the hall sensor so that the second slow approach can be executed.

paukstelis, 2024-02-22:
That seemed to fix it. Thank you for the input!

closed 2024-02-22 by paukstelis