grblHAL/core #452: Backlash Compensation cause position drift

nickshl, 2024-02-29:
Looks like in some cases backlash compensation can cause position drift. My machine has imperial balls crew(_5.08 mm or 1/5 of an inch_), my motor set to 4000 micro steps and connected directly to ball screw. So, one micro step is 0.00127 mm. I also enabled backlash compensation. Looks like in some cases mismatch between real position and grblHAL position happens.

My theory it happens because backlash compensation distance is "logic" value, not a number of steps. And this distance processed as regular line. As result number of steps per backlash movement can vary. And since those steps ignored it can cause some drift.

terjeio, 2024-03-01:
Do you have some numbers to illustrate this? E.g. what is the largest drift obsrved.

I suspect it is related to the finite resolution of single precision. The same issue that manifests itself up when using the LightBurn cluster plugin where lots of small relative movements are added to the current position.
It seems the only solution to this is to change the codebase to use `double` instead of `float`.

nickshl, 2024-03-01:
I have tool setter probe. I ran this program:
```
G38.2Z-5F50
G38.4Z5F50
G0Z10
```
copy-pasted 30 times. This is the data(_only for G38.2 command_):
```
[PRB:0.000,0.000,-7.026:1]
[PRB:0.000,0.000,-7.027:1]
[PRB:0.000,0.000,-7.028:1]
[PRB:0.000,0.000,-7.031:1]
[PRB:0.000,0.000,-7.031:1]
[PRB:0.000,0.000,-7.033:1]
[PRB:0.000,0.000,-7.035:1]
[PRB:0.000,0.000,-7.036:1]
[PRB:0.000,0.000,-7.037:1]
[PRB:0.000,0.000,-7.038:1]
[PRB:0.000,0.000,-7.041:1]
[PRB:0.000,0.000,-7.042:1]
[PRB:0.000,0.000,-7.043:1]
[PRB:0.000,0.000,-7.045:1]
[PRB:0.000,0.000,-7.045:1]
[PRB:0.000,0.000,-7.046:1]
[PRB:0.000,0.000,-7.049:1]
[PRB:0.000,0.000,-7.049:1]
[PRB:0.000,0.000,-7.050:1]
[PRB:0.000,0.000,-7.051:1]
<Idle|MPos:0.000,0.000,-3.801|Bf:100,1024|FS:0,0>
```
As you can see it rises about 1 um each time. It not always happens, looks like start position matter. Without backlash compensation difference between lowest and highest value is 2 um(!) which is surprised me since my microstep is 1.27 um.

I suspect happens something similar like this(_step is 10, backlash is 7, current positions is 0_):

G0X4
- backlash calculates new position, which is 7, less than ten, but greater than 5, so make a step, which is not counted for step position
- logic position 0, 4 less then 5, so no step, logic position changed to 4, step position remain 0

G0X-4
- backlash calculates new position, which is -3. Since step position is 0, and -5 less than -3, no backlash compensation step is happened.
-  logic position 4 changed to 0, no physical step happens

Each time this sequence happens, it will loose one step. May be changing backlash distance to be multiple of microstep distance will help... I have to try it.

nickshl, 2024-03-04:
> **P.S.** By the way, could you check Web Builder? I tried to build hex for my controller, and instead ver 20240228 it shows 20240204. Still respond to pulling TX line low and did not respond to command to gain control.

Now it shows right version, but MPG UART doesn't work at all... looks like controller TX line has low state, and MPG doesn't get anything even when sender in control.

terjeio, 2024-03-05:
I have added a quick fix to the Web Builder - select _Test2_ in the new MPG & DRO dropdown in the _Plugins tab_. Or _Test1_ and `UART` keypad...
Later I'll try to figure out how to present the options in the UI in a better way that also checks the board for available resources (I2C, UART, mode pin)...

nickshl, 2024-03-06:
> I have added a quick fix to the Web Builder - select _Test2_ in the new MPG & DRO dropdown in the _Plugins tab_.

That works! Thank you!

nickshl, 2024-06-10:
> May be changing backlash distance to be multiple of microstep distance will help... I have to try it.

I tried to set it and... my step size if 0.00127 mm, let's say I want 35 steps, it is 0.04445 mm. But when I tried to set it $161, I got only 0.044 back. May be it is just rounding during reporting, but I don't feel comfortable about that.

terjeio, 2024-06-17:
I'll increase the reported number of decimal places to 5 in the next commit.