grblHAL/core #457: Custom board with STM32F446RET6 and my issues with configuration

PecaJosef, 2024-03-03:
I have made a custom board with STM32F446RET6 and since I have worked only with grbl on ATmega328p I am getting lost in this configuration.

Unfortunately, I didn't think of an external EEPROM, so my first problem is configuring grblHAL to work without the EEPROM and using Flash for user settings instead. I really hope it is possible because the board costs me a lot of time and money.

PecaJosef, 2024-03-04:
I have found the problem. I do not know why but when selected Release STM32F446 it built it with the STM32F446RETX_BL64K_NONVS_FLASH.ld file which does not have Emulated EEPROM. I do not know how to fix it elegantly so I have copied the STM32F446RETX_FLASH.ld content to the STM32F446RETX_BL64K_NONVS_FLASH.ld and it worked.

The only problem I have right now is that I am unable to turn on the spindle and set its speed :D.

samitray, 2024-05-01:
It depends on the value of _EEPROM_Emul_Sector supplied in the linker file. So depending on which part of flash was used for emulation, the value of _EEPROM_Emul_Sector has to be changed as per sector numbers in the data sheet
see Discussion -> General  #503