grblHAL/core #458: Qestion about min/max values

shaise, 2024-03-05:
Hi @terjeio 

I'm in a process now to adapt my sender to use your internal parameter definition report instead of a fixed one. Thanks for this great feature!
I have a question though about parameter 481: autoreport.
the default value is 0, (disable) but the min/max allowed values are 100 to 1000. This causes an issue in my implementation as I do not allow values lower then min, hence I can not disable autoreport unless I do a special 'if' for this parameter.

what do you suggest? How do you handle it?
Thanks!

shai

terjeio, 2024-03-05:
There is a flag at the end of the settings report which you can check - if 1 blank (for strings) or 0 is allowed for the setting.
```
[SETTING:481|1|Autoreport interval|ms|5|###0|100|1000|1|1]
                                                        ^
```

shaise, 2024-03-05:
Ahhh! NICE!!!

Thanks!

shaise, 2024-03-06:
Thanks again!
Works great now.
[image]

closed 2024-03-06 by shaise