grblHAL/core #464: Clear TLO reference on homing cannot be handled by plugins

rcp1, 2024-03-16:
Hi,

if I would want to takeover the handling of TLO / TLR in a plugin, I can't clear their state on homing in the moment since the current iface namely `tc_clear_tlo_reference` is hard-coded in the homing cycle:

https://github.com/grblHAL/core/blob/06f442643a115ab3acd6722b4afc1fcee8d9d7be/motion_control.c#L833

Would it be possible to add a `grbl.` function for this, so I am able to inject plugin code here?

terjeio, 2024-03-18:
I wonder if it will be better to hook `tc_clear_tlo_reference` into the `grbl.home_machine` handler chain and only do so if the core tool change code is active? And move `tc_clear_tlo_reference` to _system.c_ and remove the mode check so that it can be reset by your code by also hooking it into the `grbl.home_machine` chain?

terjeio, 2024-03-18:
I am going to change the signature of `grbl.on_homing_complete` (add the cycle mask) and `move tc_clear_tlo_reference` to _system.c_ as `system_clear_tlo_reference` (without the mode check).

rcp1, 2024-03-18:
Sounds good, one could even hard-code `system_clear_tlo_reference` to always execute in homing of the tool change axis.

terjeio, 2024-03-19:
FYI the latest commit contains the change.

hanke-cnc, 2024-03-20:
> FYI the latest commit contains the change.

"Hi, terjeio, the Chinese translation file I submitted for ioSender hasn't been accepted."

https://github.com/terjeio/ioSender/issues/366

closed 2024-03-30 by terjeio