grblHAL/core #468: Dust Collection Pin Invert?

Donagin, 2024-03-22:
I am using Phil Barrett's Teensy 4.1 BOB, and controlling outlets for dust collector, flood coolant, and etc on my main power box.  Using the common cheap 4 Channel DC 5V Relay Module (ELLEGOO or similar from China), I want to use the NO sides on the relays so that if the teensy is offline while the 5 volt power is online, the outlets controlled by the relays remain off.  I have been able to invert all of the relay signals to the board so that it behaves as desired - except the dust collector.  There does not appear to be any option to invert this pin.  I have been wrong before - so if I am missing something I apologize.  How hard would this be to add correctly?

terjeio, 2024-03-25:
IIRC the dust collector enable signal is hardwired to spindle enable so cannot be independently inverted. @phil-barrett can confirm this.

phil-barrett, 2024-03-25:
The DC relay mirrors the Spindle relay, so no you can't invert it.  

However, I do not understand what you want to do.  When you say "teensy offline", do you mean unpowered?  If that is the case, then the relay would be off. The DC relay TTL signal will only be high when the Spindle relay TTL signal is high. When Teensy is unpowered, the Spindle relay signal is low. (Those are the pin header outputs, by the way.)

By the way, the screw terminal outputs are a little different to use.  They are for relay coils, not powered relay boards like you are using.

For more information about using relays, [read here]([url](https://www.grbl.org/single-post/using-relays-with-grblhal-and-the-t41u5xbb)).

closed 2024-04-16 by terjeio