grblHAL/core #471: Wrong spindle mask?

Nick507, 2024-03-25:
In my_machine.h:
`#define SPINDLE0_ENABLE           SPINDLE_VFD`
I guess it should be:
`#define SPINDLE0_ENABLE           (1<<SPINDLE_MODVFD)`

and in driver_opts.h:274
`#define SPINDLE_ENABLE ((1<<SPINDLE0_ENABLE)|(1<<SPINDLE1_ENABLE)|(1<<SPINDLE2_ENABLE)|(1<<SPINDLE3_ENABLE))`
should be:
`#define SPINDLE_ENABLE ((SPINDLE0_ENABLE)|(SPINDLE1_ENABLE)|(SPINDLE2_ENABLE)|(SPINDLE3_ENABLE))`

Because as I understand, you want to get intersection mask between all types of used spindles.

terjeio, 2024-03-26:
`#define SPINDLE0_ENABLE           SPINDLE_VFD`
is not correct since the `SPINDLE_VFD` symbol is not defined anywhere, use one of the [predefined symbols or values](https://github.com/grblHAL/core/blob/7cf86877f0a35f7881eb971d495acdc29103660d/spindle_control.h#L29-L51).

> Because as I understand, you want to get intersection mask between all types of used spindles.

Correct, and I chose to bit shift the spindle number in _driver_opts.h_.

Nick507, 2024-03-26:
> is not correct since the SPINDLE_VFD symbol is not defined anywhere, use one of the [predefined symbols or values](https://github.com/grblHAL/core/blob/7cf86877f0a35f7881eb971d495acdc29103660d/spindle_control.h#L29-L51).
The mistake here not just SPINDLE_VFD, but value must be a bit mask (1<<X) whereas current code is:
`//#define SPINDLE0_ENABLE         SPINDLE_HUANYANG1`

actually this comment relates to ESP32 driver repo, I've just add it here, because it is related with bug in driver_opts, which is core repo.

Nick507, 2024-03-26:
There is no issue

closed 2024-03-26 by Nick507