grblHAL/core #472: spindle_get_caps() hardfault after NVS reset

dresco, 2024-03-26:
While chasing down a crash, I've noticed `spindle_get_caps()` being called while `n_spindle` is set to 0. This only seems to happen after NVS reset, and the call chain is `nvs_buffer_init() -> settings_restore() -> spindle_get_caps()`. Not sure whether that is expected, or means there is something wrong with my board map etc..?

Anyway, this ends up in the hard fault handler, as `idx` is decremented from zero before `spindles[idx]` is referenced, and points at an invalid memory address around 4GB (seems the uint_fast8_t type is actually an unsigned 32 bit integer on STM32 platform).

dresco, 2024-03-27:
Thanks for the quick fix Terje..

I've noticed that the spindle pins **must** be configured via AUXOUTPUT ports now (else the PWM timer doesn't get set up and spindle initialisation fails).

Is that the intent going forwards? Just wanting to check before updating my map files..

terjeio, 2024-03-27:
> I've noticed that the spindle pins must be configured via AUXOUTPUT port

Yes, it is a change I am making to free up the PWM spindle pin(s) as aux pin(s) when PWM spindle is not configured or only PWM and on/off is output required. The coolant pins will likely be configured in the same way later.

dresco, 2024-03-27:
Cool, thanks..

closed 2024-03-27 by dresco