grblHAL/core #480: Axis remapping?

paukstelis, 2024-04-01:
Is it currently possible in GrblHal to either remap an axis definition (e.g. A  to C)? The issue I am running into is that the LPC17xx board I am using has a maximum of 5 axes, and it seems that GrblHal requires that these be defined in the order X,Y,Z,A,B,C (at least the way I understand it). I would like to use X,Z,B,C, but I cannot define a C axis since it expects there to be a Y axis and then A and B first.

terjeio, 2024-04-03:
You can by modifying the core [here](https://github.com/grblHAL/core/blob/c3aace735bdc4e11e256d94a6a78c7810d30000c/gcode.c#L1327).
I have no plans to add axis remapping like this as standard as it will affect too many parts of the code, and likely senders as well.

paukstelis, 2024-04-03:
Thanks. That will do.

closed 2024-04-03 by paukstelis