grblHAL/core #482: GrblHAL Web Builder Build failed: 422 - SAM3X8E

PolyEng, 2024-04-02:
Hi, new to the whole issue thing on github so if im in the wrong place feel free to scold me.

The problem:

using the grblhal web builder and building release 20240330 for the SAM3X8E (arduino due) for the RAMPS 1.6 i get Build failed: 422

All settings are the defaut ones i **only added the tick on** >Advanced features >**RS274 NGC expression support**, as far as i undestand RS274 NCG is needed for probing (as now when issuing G38.3 through IOsender machine says invalid gcode , g code generated from iosender built in probing tab)

this is the output error from grblhal web builder:

[SAM3X8E_RAMPS_16.json](https://github.com/grblHAL/core/files/14843003/SAM3X8E_RAMPS_16.json)

Command failed: pio run -etest -cjob.ini
src/driver.c:300:24: warning: 'probe_pin' defined but not used [-Wunused-variable]
static input_signal_t *probe_pin;
^~~~~~~~~
In file included from src/driver.c:32:0:
src/grbl/motor_pins.h:1270:13: warning: 'motor_iterator' defined but not used [-Wunused-function]
static void motor_iterator (motor_iterator_callback_ptr callback)
^~~~~~~~~~~~~~
src/grbl/spindle_control.c:54:22: warning: 'spindle_cap' defined but not used [-Wunused-variable]
static spindle_cap_t spindle_cap;
^~~~~~~~~~~
/home/terjeio/.platformio/packages/framework-arduino-sam/cores/arduino/USB/USBCore.cpp: In function 'void USB_ISR()':
/home/terjeio/.platformio/packages/framework-arduino-sam/cores/arduino/USB/USBCore.cpp:688:21: warning: this 'else' clause does not guard... [-Wmisleading-indentation]
else
^~~~
/home/terjeio/.platformio/packages/framework-arduino-sam/cores/arduino/USB/USBCore.cpp:690:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'else'
UDD_Send8(EP0, 0);
^~~~~~~~~
.pio/build/test/src/grbl/ngc_flowctrl.c.o: In function `ngc_flowctrl':
/home/terjeio/grblHAL/SAM3X8E/grblHAL_Due/src/grbl/ngc_flowctrl.c:264: undefined reference to `vfs_tell'
/home/terjeio/grblHAL/SAM3X8E/grblHAL_Due/src/grbl/ngc_flowctrl.c:379: undefined reference to `vfs_seek'
/home/terjeio/grblHAL/SAM3X8E/grblHAL_Due/src/grbl/ngc_flowctrl.c:290: undefined reference to `vfs_tell'
/home/terjeio/grblHAL/SAM3X8E/grblHAL_Due/src/grbl/ngc_flowctrl.c:307: undefined reference to `vfs_seek'
/home/terjeio/grblHAL/SAM3X8E/grblHAL_Due/src/grbl/ngc_flowctrl.c:324: undefined reference to `vfs_tell'
/home/terjeio/grblHAL/SAM3X8E/grblHAL_Due/src/grbl/ngc_flowctrl.c:387: undefined reference to `vfs_seek'
collect2: error: ld returned 1 exit status
*** [.pio/build/test/firmware.elf] Error 1
========================= [FAILED] Took 21.19 seconds =========================

terjeio, 2024-04-02:
My bad, the driver did not handle exough aux inputs to accomodate the probe. Should be fixed now.
FYI the RS274 NCG is not needed for probing - and the compilation failure was due to the Arduino framework (?) defining a lower case symbol that shomehow messed up my code. After undefining the offending symbol the RS274 NCG option compiles and should work.

PolyEng, 2024-04-03:
This is the message I get when I try to run g38.3 on my board with RS274 NCG NOT enabled, 
[image]
[image]
 The noprobe thing I suppose it's because probe isn't enabled, even if probe input is enabled by default in the web builder

terjeio, 2024-04-03:
Build anew - I committed a fix yesterday. Or have you already done so?

PolyEng, 2024-04-03:
@terjeio is there any way to support you for your wonderful work?

terjeio, 2024-04-04:
I have now updated the board "drawing" to match the board map.

Currently I do not accept donations since EU(?) regulations requires me to declare donations beforehand, even which country they might come from - and since I am not able to look into the future how can I do that? If I do accept donations anyway I guess I may lose access to banking services, but not I am not sure since this is not stipulated in the form I have to fill out now and then...