grblHAL/core #485: Macro is giving error about not running from filesystem when it is.

rvalotta, 2024-04-06:
Taking a stab at wriring a ATC macro that uses macros as functions...  I 'm getting an odd error and I can't seem to find what I may be doing wrong.

The Macro is as follows

```
; #7 - Pocket 1 X pos
; #8 - Pocket 1 Y Pos
; #3 - Pocket 1 Alignment 0=X 1=Y
; #11 - Pocket Direction 0=Positive 1=Negative
; #4 - Pocket Offset
; #5 - Tool Number
; #6 - Axis to return 0=X 1=Y

(debug, starting tool pos)
#31 = [#5 - 1]
#32 = [#31 * #4]
(debug, #5, #31, offset is #32)

; X Align
o100 if [#3 EQ 0] 
(debug, ATC Aligned with X-Axis)
; Positive
    o110 if [#11 EQ 0] 
        (debug, Pocket 1 is on the left)
        o111 if [#6 EQ 0] 
            #33 = [#7 + #32]
            (debug, X Pos is #33)
            o111 return [#33]
        o111 else 
            (debug, y Pos is #8)
            M99
            o111 return [#8]
        o111 endif
    o110 else 
        (debug, Pockets 1 is on the right)
        o112 if [#6 EQ 0] 
            #33 = [#7 - #32] 
            (debug, x pos is #33)
            o112 return [#33]
        o112 else 
            (debug, y pos is #8)
            o112 return [#8]
        o112 endif 
    o110 endif
o100 else 
(debug, ATC Aligned in Y-Axis)
    o120 if [#11 eq 0] 
        (debug, Pocket 1 is on the bottom)
        o121 if [#6 EQ 0] 
            (debug, X pos is #7)
            o121 return [#7]
        o121 else 
            #33 = [#8 + #32]
            (debug, Y pos is #33)
            o121 return [#33]
        o121 endif
    o120 else 
        (debug, Pocket 1 is on the top)
        o122 if [#6 EQ 0] 
            (debug, X pos is #7)
            o122 return [#7]
        o122 else 
            #33 = [#8 - #32]
            (debug, Y pos is #33)
            o122 return [#33]
        o122 endif   
    o120 endif
o100 endif
```

When called as follows G65 P100 D100 E100 F0 H0 I30 J2 K0 It works as expected while G65 P100 D100 E100 F0 H0 I30 J2 K1 gives a error 80 that the script is not running from the filesystem.

With the debugging I'm getting to  
[MSG:starting tool pos]
[MSG:2.000000, 1.000000, offset is 30.000000]
[MSG:ATC Aligned with X-Axis]
[MSG:Pocket 1 is on the left]

I should be seeing Y Pos is... any help would be appreciated...

terjeio, 2024-04-08:
Note that the M99 in the O111 else block will terminate the macro - the following return statement will never be executed.

rvalotta, 2024-04-09:
I am aware thank you I was trying to determine what line was causing me grief and added that in to see if it was the return or something else.

closed 2024-04-09 by rvalotta