grblHAL/core #487: Need Help to set up grblHAL

timeO25, 2024-04-08:
Hello,
I have an Teensy 4.1 and like to run grblHAL on it with ethernet.
Can someone tell me what all the $$ configurations do and maybe give me an example how to send a G-Code with python to the T41U5XBB via ethernet and give me the correct hex file for that to load on the Teensy 4.1.
Thank you very much.
Best regards.

terjeio, 2024-04-08:
Build your hex with the [Web Builder](http://svn.io-engineering.com:8080/?driver=iMXRT1062&board=T41U5XBB). Enable networking and telnet server in the _Network/WebUI_ tab.
Default is that the IP address will be assigned with DHCP, `$I` output (in a terminal) will contain the address.

Somebody else has to provide you with python code - or you have to change the code provided in the [Grbl repo](https://github.com/gnea/grbl/tree/master/doc/script).

[ioSender](https://github.com/terjeio/ioSender/releases/tag/2.0.44) (or the config app) has a config UI with description of all settings. Or see the [Wiki](https://github.com/grblHAL/core/wiki/Additional-or-extended-settings).

timeO25, 2024-04-08:
Thanks for your fast help.
When I typed $I in the terminal there was no IP in the output only IP:
And do you have a list of all $... when using the $$ command so that I know what they do/change.

terjeio, 2024-04-08:
Reset the settings with `$RST=*` and turn the board off and on again. Working now?

> And do you have a list of all "$..." when using the "$$" command so that I know what they do/change.

Yes that is build into the controller, in a terminal send `$help settings` to get the full list. Or just `$help` to get the topics you can use with the `$help` command. Again, this is described in the [Wiki page linked above](https://github.com/grblHAL/core/wiki/Additional-or-extended-settings).

Some things to check:

Is there light in the green LED in the ethernet socket? Blinking occasionally?
Does the router you are connected to support DHCP? If not you need to change to a static IP address, change $301 - $304 to match your network.
Is the 6 pin ethernet header soldered in the Teensy board?.

timeO25, 2024-04-09:
Does the T41U5XBB need an external power suply for the ethernet adapter, because der is no light blinking.

terjeio, 2024-04-09:
No external supply is needed for ethernet. The 6-pin header is to be soldered in the red rectangle:

[image]

and a matching socket has to be added to the breakout board if not present. See the [User Manual](https://github.com/phil-barrett/grblHAL-teensy-4.x/blob/master/T41U5XBB%20v207.pdf) for details.

timeO25, 2024-04-09:
Thank you very much these pins where missing now it works thanks.

timeO25, 2024-04-09:
Do you know where I can ask to get code to send the G-Code via Network by my own with Java?

karoria, 2024-04-09:
The readily available senders/webUI for grblHAL:

1. IO Sender (C#)
2. G-Sender (JS)
3. Standard webui by Luc (JS)
4. GrblTouch webui by me (JS)

There might be others too but not known to me for compatibility with grblHAL.

timeO25, 2024-04-09:
Thank you very much I'll try that.

terjeio, 2024-04-09:
> Do you know where I can ask to get code to send the G-Code via Network by my own with Java?

Search internet for "java telnet client" - that may give you some pointers.

FYI UGS is a sender that is written in java, bCNC is written in python.

timeO25, 2024-04-10:
Thank you very much for your help I'll try that.

closed 2024-04-16 by terjeio