grblHAL/core #493: Not an issue: Want to know if macro variables are available for $342 to $344.

karoria, 2024-04-22:
Hi @terjeio 
I want to know if macro variables are available for some useful settings like $342 to $344 so that I can use them in tc.macro. There might be other settings too which if served as variables, can be really useful.
Cheers!
Ravi

terjeio, 2024-04-23:
Currently not, perhaps they could be made available via a "standard" G65 taking a parameter for the setting number? Should be possible if nesting works...

karoria, 2024-04-23:
That's great idea. I will wait for it if you plan to go ahead.
Thanks.

terjeio, 2024-04-25:
I have committed a [tentative solution](https://github.com/grblHAL/core/blob/master/changelog.md), can you verify? The setting value is returned in `_value` and `_value_returned` is set to 1 if the setting is available and of a numeric type.