grblHAL/core #498: Joggin unavailable for many commands

nickshl, 2024-04-25:
The main thing about jogging that "_Jogging command do not alter the g-code parser state in any way_". Unfortunately jogging acts as G1 command. It would be really nice if jogging will support G0(rapid movement), G2/G3(circular interpolation), G38(probing), G43(set tool offset) and may be some others. Is it any chance it would be implemented?

terjeio, 2024-04-25:
You want that so that the mentioned G-codes does not change the parser state? Adding M70/M72 support will be a better alternative if so?

nickshl, 2024-04-25:
If it would be possible to use it in a single line like this:
> M70<some different stuff that changes modal state and does the movement>M71\r\n

Then it would work.