grblHAL/core #501: kinda vfd support

thirstyone, 2024-04-25:
When I look here https://github.com/grblHAL/drivers?tab=readme-ov-file -  specifically, at spindle modbus support line - there's a bit of confusion for me. Is it done artificially, or is there any fundamental reason for a firmware for an stm32f407 or stm32f411 not to support modbus spindle control? 

Then, there's another question - on choosing the right option in web-configurator. Initially I thought they (vfds) are using the same application-level protocol,   so to say, and a common instruction set. But after looking through that part of code it looks like implementation is different for each make. I am using the KD200-00220S2 from Kinda. Does it mean, I have to provide a pull request to get it supported by firmware?

terjeio, 2024-04-25:
>  is there any fundamental reason for a firmware for an stm32f407 or stm32f411 not to support modbus spindle control?

No, the difference is if a specific board has support for the UART (serial) port required.

>  it looks like implementation is different for each make.

More or less so, if there is a standard many do not adhere to it.

> Does it mean, I have to provide a pull request to get it supported by firmware?

If you can use the MODVFD driver, which is configurable via settings, then no. If not a new driver has to be added.

thirstyone, 2024-04-28:
Thanks! Is there any manual/tutorials on how to configure MODVFD? Settings - you mean, recompile firmware or those $462-$471 parameters listed on https://github.com/grblHAL/Plugins_spindle ? 
From what I see in the user manual for my VFD, it looks different, doesn't it? 
[Kinda - KD100 and KD200 user manual.pdf](https://github.com/grblHAL/core/files/15143251/Kinda.-.KD100.and.KD200.user.manual.pdf)

terjeio, 2024-04-29:
> Is there any manual/tutorials on how to configure MODVFD?

Not that I am aware of, except that it has been adapted from LinuxCNC so [this thread](https://forum.linuxcnc.org/24-hal-components/38733-vfdmod-an-easy-vfd-control-over-modbus-rtu) may cointain some pointers.

> From what I see in the user manual for my VFD, it looks different, doesn't it?

I think MODVFD could be used since command codes 3 and 6 are used, settings $462-$471 has to be set up to match the VFD.