grblHAL/core #504: Macro Issue that I can not figure out.

rvalotta, 2024-04-29:
We are continuing to work on the ATC macro that I reported issues with last time and am experiencing an odd issue.

The following block of code is behaving oddly..   

`o400 if [#<_selected_tool> EQ 0]
  ; We selected tool 0. Do nothing as we are already loaded.
  (debug, Loaded tool 0)
o400 elseif [#<_selected_tool> LE #<_rc_pockets>]
  ; We have a tool with a pocket
  G53 G0 X[#<_rc_x_load>] Y[#<_rc_y_load>]
  (debug, Move to pocket #<_selected_tool>)
  G53 G0 Z[#<_rc_z_engage> + #<_rc_z_spin_off>]
  (debug, Move to spin start)
  G53 G1 Z[#<_rc_z_spin_load>] F[#<_rc_feed_engage>]
  M3 S[#<_rc_rpm_load>]
  (debug, Run spindle CW)

  #<_rc_plunged_count> = 0
  o410 while [#<_rc_plunged_count> LT #<_rc_plunge_count>]
    G53 G1 Z[#<_rc_z_engage>] F[#<_rc_feed_engage>]
    (debug, Engage)
    G53 G1 Z[#<_rc_z_engage> + #<_rc_z_retreat_off>] F[#<_rc_feed_engage>]
    (debug, Retreat)
    #<_rc_plunged_count> = [#<_rc_plunged_count> + 1]
    (debug, Load plunge #<_rc_plunged_count> complete)
  o410 endwhile

  M5
  (debug, Stop spindle)
  G53 G0 Z[#<_rc_z_to_measure>]
  (debug, Move to measure z)
  (debug, Confirm tool #<_selected_tool> is loaded and press cycle start to continue.)
  M0
o400 else
  (debug, Came into o400 else)
  ; Tool out of magazine range. Load manually
  G53 G0 X[#<_rc_x_manual>] Y[#<_rc_y_manual>]
  (debug, Tool #<_selected_tool> is out of magazine range. Manually load tool #<_selected_tool> and cycle start to continue.)
  M0
  (debug, Loaded tool out of range)
o400 endif
`

When it is suppose to go into the o400 else section we are getting the following error
[MSG:Warning: error 81 in macro P99.macro]
error:81 - Unknown flow statement.

I don't see anything wrong with any of the flow statements I'm at a loss as to why this is not working.  We are not even getting the debug message saying it has made it to the else.

Thanks for any help in advance.

terjeio, 2024-04-29:
more bugs...
Are you compiling locally or using the Web Builder?

rvalotta, 2024-04-29:
Precompiled version provided for the flexihal...  But I can compile manually if needed.   Was going to try and add a option to echo the line of code it is failing on to help better track down what might be wrong

terjeio, 2024-04-29:
Ok, here is the patched code (in the grbl folder):
[ngc_flowctrl.zip](https://github.com/grblHAL/core/files/15154437/ngc_flowctrl.zip)
I have added the echo.

rvalotta, 2024-04-30:
Thank you that Fixed the issue...

rvalotta, 2024-05-03:
Just out of curiosity any idea when this might make it into the firmware?   Would like to make the script available to others but can't do that until this bug is fixed and available for others to use. 

Thanks in advance

andrewmarles, 2024-05-03:
FWIW I published a preview build for FlexiHAL here:
https://github.com/Expatria-Technologies/STM32F4xx/releases/tag/flexi-hal-v1.0.0.1

closed 2024-06-25 by terjeio