grblHAL/core #510: Is the data in the driver capabilities list dependably up to date?

imechura, 2024-05-09:
Hi,

I am wondering if the driuver capability matrix found here https://github.com/grblHAL/drivers is up to date. 

The reason I am asking is because I see it says that Digital Aux I/O is not supported so I prepared to migrate my project over to RPI Pico which will be a substantial amount of work but then I noticed in some of the board maps for ESP32 there is, what I beleive to be configuration for AUX pins as can be on line 93 here:
https://github.com/grblHAL/ESP32/blob/master/main/boards/generic_map.h

Perhaps this configuration is for something else?

Thanks,
Ian

terjeio, 2024-05-09:
> I am wondering if the driuver capability matrix found here https://github.com/grblHAL/drivers is up to date.

Sorry, it is not...

Aux [inputs](https://github.com/grblHAL/ESP32/blob/804d3af65c70c187f925c2a33ff69cff15c379dc/main/driver.c#L247-L267) and [outputs](https://github.com/grblHAL/ESP32/blob/804d3af65c70c187f925c2a33ff69cff15c379dc/main/driver.c#L396-L431) can be added to the board map if there are free pins. Some of these may be "claimed" for optional pins that has specific functionality - as seen in the generic map linked above.

imechura, 2024-05-10:
Thats what I needed to know. Thank you.

closed 2024-05-10 by imechura