grblHAL/core #512: set machine origin to 0, or  bit3, not working for $22

terjeio, 2024-05-13:
### Discussed in https://github.com/grblHAL/core/discussions/511

<div type='discussions-op-text'>

<sup>Originally posted by **bertabus** May 12, 2024</sup>
My homing is bottom left, it's a plotter, I'm trying to get the last few settings to work and not having luck.
When I set `$22=15` and then check the setting with `$22` the result is `7`.
When I set `$22=11` and then check the setting with `$22` the result is `3`.
No matter what I set the `$22` bitmask to, grblHAL zero's the bit3 mask.
What am I missing to be able to set bit3 - "set machine origin to 0" and zero out my machine on homing?

NOTE: the work coordinates do update with homing, but not the machine coordinates.
Machine position after homing shows X:-209 Y:-279, which appears to be the limits minus the homing backoff.
`G28.3 X0` doesn't seem to be valid with grblHAL, so not sure how to work around this.

Settings:
```
> $$
$0=10.0 (Step pulse time, microseconds)
$1=25 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=5 (Step direction invert, mask)
$4=7 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$9=1
$10=511 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$14=70
$15=0
$16=0
$17=0
$18=0
$20=0 (Soft limits enable, boolean)
$21=1 (Hard limits enable, boolean)
$22=7 (Homing cycle enable, boolean)
$23=3 (Homing direction invert, mask)
$24=25.0 (Homing locate feed rate, mm/min)
$25=3000.0 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, milliseconds)
$27=1.000 (Homing switch pull-off distance, millimeters)
$28=0.100
$29=0.0
$30=1000.000 (Maximum spindle speed, RPM)
$31=1.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$33=750.0
$34=0.0
$35=0.0
$36=100.0
$37=0
$39=1
$40=0
$43=1
$44=3
$45=0
$46=0
$62=0
$63=3
$64=0
$100=80.00000 (X-axis travel resolution, step/mm)
$101=80.00000 (Y-axis travel resolution, step/mm)
$102=480.00000 (Z-axis travel resolution, step/mm)
$110=5000.000 (X-axis maximum rate, mm/min)
$111=5000.000 (Y-axis maximum rate, mm/min)
$112=1000.000 (Z-axis maximum rate, mm/min)
$120=500.000 (X-axis acceleration, mm/sec^2)
$121=500.000 (Y-axis acceleration, mm/sec^2)
$122=100.000 (Z-axis acceleration, mm/sec^2)
$130=210.000 (X-axis maximum travel, millimeters)
$131=280.000 (Y-axis maximum travel, millimeters)
$132=40.000 (Z-axis maximum travel, millimeters)
$140=500
$141=500
$142=500
$150=16
$151=16
$152=16
$180=25.0
$181=25.0
$182=25.0
$190=500.0
$191=500.0
$192=500.0
$200=22.0
$201=22.0
$202=218.0
$210=50
$211=50
$212=50
$220=22.0
$221=22.0
$222=22.0
$339=0
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=200.0
$346=1
$370=0
$384=0
$394=4.0
$398=100
$481=0
$484=1
$486=0
$673=1.0
ok
```

```
> $i
[VER:1.1f.20240506:]
[OPT:VNHS,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,NOPROBE,ES,SED,RTC,TMC=7]
[FIRMWARE:grblHAL]
[SIGNALS:HSE]
[NVS STORAGE:*FLASH]
[FREE MEMORY:217K]
[DRIVER:RP2040]
[DRIVER VERSION:240408]
[DRIVER OPTIONS:SDK_1.5.0]
[BOARD:BTT SKR Pico 1.0]
[AUX IO:2,0,0,0]
[PLUGIN:Trinamic v0.16]
``` 

Any help or tips appreciated.</div>

Flowbey, 2024-07-04:
i have the same issue but the fix didn't work for me. still can't set $22=9.

terjeio, 2024-07-04:
> i have the same issue but the fix didn't work for me. still can't set $22=9.

Compatibility level is set > 1? Post your `$I` output.

Flowbey, 2024-07-05:
I just compiled now with compatibility level 1 but still the same

[VER:1.1f.20240420:]
[OPT:VNMDSL,100,1024,4,0]
[AXS:4:XYZA]
[NEWOPT:ENUMS,RT+,HOME,NOPROBE,ES,SED,TMC=15,ODO,SD]
[FIRMWARE:grblHAL]
[SIGNALS:HSE]
[NVS STORAGE:*EEPROM 4K]
[FREE MEMORY:86K]
[DRIVER:STM32F446]
[DRIVER VERSION:240428]
[BOARD:BTT Octopus V1.1]
[AUX IO:6,3,0,0]
[PLUGIN:Bootloader Entry v0.02]
[PLUGIN:Trinamic v0.16]
[PLUGIN:Macro plugin v0.06]
[PLUGIN:Fans v0.13]
[FANS:1]
[PLUGIN:ODOMETERS v0.06]
[PLUGIN:SDCARD v1.13]
[PLUGIN:FS macro plugin v0.07]

terjeio, 2024-07-05:
> [VER:1.1f.20240420:]

The fix was comitted in build [20240513](https://github.com/grblHAL/core/blob/master/changelog.md#20240513) so update your source.

Flowbey, 2024-07-05:
First i tried change settings.c directly with your fix to "homing.value = settings.homing.flags.value & 0b1111;"
After many Hours i figured out, that my platformio was not using the right source to compile. Strange behavior happening.
Now it works. Your input with the version directed me to the solution. 
Thank you again.