grblHAL/core #517: Not correctly reporting coordinates?

tcurdt, 2024-05-19:
Machine is homed. I moved to a different position and the set to zero.

This should have set the working coordinates P1 to zero but the DRO still shows the machine coordinates.

[image]

Now it seems like the machine still does the right thing. `G0 Z10` and `G0 Z0` does work correctly.
So I thought maybe it's a problem with the sender - but UGS reported the same. A soft reset did not change things either.

When I restarting the sender I noticed

```
Connected to /dev/tty.usbmodem325B386534321 with a baud rate of 115200
client ?
client $$
<Idle|MPos:170.000,60.000,-77.500|Bf:128,1024>
error:2 (Bad number format)
```

Does the sender maybe expect a different format?
What about the working coordinates - not just the machine coordinates?
What's Bf? 

I am a little lost how to track down whether this is a sender problem.
Is there any other way to get the full DRO coordinates from grblHAL to check?

tcurdt, 2024-05-19:
Feel free to turn this into a discussion if more appropriate. I could not make out whether this is a bug or whether I am just holding it wrong.

terjeio, 2024-05-19:
Looks like you have imported settings from Grbl - not all are compatible, notably $10 that has way more options than Grbl. Default value for $10 is 511 which enables quite a bit more data in the real-time report than shown above. So try $10=511 as a start. It could also be the sender settting $10 on connection, which means you have to run the controller in a [compatibility mode](https://github.com/grblHAL/core/wiki/Compatibility-level).
Tip: use `$$=10` to output the available options or check the [Wiki page](https://github.com/grblHAL/core/wiki/Additional-or-extended-settings).

The error 2 message is from the controller, from receiving something it did not understand. Not sure what that could be.

> What's Bf?

Free space in the planner buffer and serial input buffer.

tcurdt, 2024-05-19:
Hmmmmm. I haven't really imported anything. But I suspect I know the culprit. I was using the gsender to edit some firmware settings. Maybe it overwrote some settings without me being aware.

Thanks for the pointers! I'll report back soon.

tcurdt, 2024-05-19:
Alright. It was gsender that switched from grblHAL to grbl during auto-connect (for whatever mysterious reason).

closed 2024-05-19 by tcurdt