grblHAL/core #521: Experiencing issues with ESP32 Blackbox Build

rvalotta, 2024-05-22:
Tried to compile in ngc support and am getting the following when i try and call a script from the sd card.

[16:31:07] [ $I ] ok
[16:31:07] [ $G ] [GC:G0 G54 G92 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
[16:31:07] [ $G ] ok
[16:31:07] [ connect ] Firmware Detected: grblHAL version 1.1f dated on COM3
[16:31:24] [ G65 P200 ] Guru Meditation Error: Core 1 panic'ed (LoadStoreError). Exception was unhandled.
[16:31:24] [ G65 P200 ] Core 1 register dump:
[16:31:24] [ G65 P200 ] PC : 0x400e7388 PS : 0x00060b30 A0 : 0x800d9658 A1 : 0x3ffb64a0
[16:31:24] [ G65 P200 ] A2 : 0x3f400f79 A3 : 0x3f400f79 A4 : 0x0000005f A5 : 0x00000020
[16:31:24] [ G65 P200 ] A6 : 0x00000019 A7 : 0x00000002 A8 : 0x0000001e A9 : 0x3ffb6698
[16:31:24] [ G65 P200 ] A10 : 0x00000000 A11 : 0x00000000 A12 : 0x00000014 A13 : 0x000000a5
[16:31:24] [ G65 P200 ] A14 : 0x000000a5 A15 : 0x00000009 SAR : 0x00000010 EXCCAUSE: 0x00000003
[16:31:24] [ G65 P200 ] EXCVADDR: 0x3f400f79 LBEG : 0x4000c46c LEND : 0x4000c477 LCOUNT : 0x00000000
[16:31:25] [ G65 P200 ] Backtrace:0x400e7385:0x3ffb64a0 0x400d9655:0x3ffb64d0 0x400dd584:0x3ffb65a0 0x400d5ae8:0x3ffb65e0 0x400d31db:0x3ffb6610
[16:31:25] [ G65 P200 ] ELF file SHA256: 388a9c6b606ae683
[16:31:25] [ G65 P200 ] CPU halted.

That is happening when I execute G65 P200

I've attached the build settings from the web builer.

[ESP32_BLACKBOX_X32 (1).json](https://github.com/grblHAL/core/files/15396757/ESP32_BLACKBOX_X32.1.json)

terjeio, 2024-05-22:
According to the backtrace it crashes when trying to set a named parameter. It does not crash if I set one manually (with a ESP32-S3).
Can you share enough of the macro so that I can try it here?

rvalotta, 2024-05-22:
We have cropped the macro down to just this and it still crashes

`; 
; Copyright (C) 2024 Greilick Industries LLC

; RapidChange ATC Macros for GrblHAL is free software:
; You can redistribute it and/or modify it under the terms
; of the GNU General Public License as published by
; the Free Software Foundation, under version 3 of the License.

; RapidChange ATC Macros for GrblHAL is distributed in the
; hope that it will be useful, but WITHOUT ANY WARRANTY
; without even the implied warranty of MERCHANTABILITY or
; FITNESS FOR A PARTICULAR PURPOSE. See the GNU General Public
; License for more details.

; You should have received a copy of the GNU General Public License
; along with RapidChange ATC Macros for GrblHAL.
; If not, see https://www.gnu.org/licenses/.
; 

; ** BEGIN USER CONFIGURATION **
; ATC Operations
; The units for your configuration: 20 = Inches, 21 = Millimeters
#<_rc_units> = 21
(debug, Units: #<_rc_units>)`

terjeio, 2024-05-22:
If you send the two command from the MDI does it still crash?
When I do so with my S3 it does not (with the patched version, but I believe that should not matter).

rvalotta, 2024-05-22:
Running 2 2 lines of gcode from the MDI worked without any issue.

terjeio, 2024-05-22:
Ok, then there is something else going on. Since I am away and do not have access to the hardware needed this has to wait until I am back home (early next month).

FYI the code ran as it should from the SD card on the STM32F7 board I have here.

terjeio, 2024-05-26:
Got hold of a ESP32 dev board so could debug - crash was due to attempting to write back to flash when lowercasing variable names.

rvalotta, 2024-05-26:
I will do a custom compile this afternoon and test it out. Thank you

rvalotta, 2024-06-07:
Saw that the web builder was updated. Tried to build a fresh copy and am getting an unable to mount sd card.   For shits and giggles I took a esp32 and sd card adapter and am getting the same thing.   Not sure what to look at to try and track it down.

terjeio, 2024-06-07:
I'll look into this early next week when I have access to the neccesary hardware.

rvalotta, 2024-06-07:
I ended up reformatting the card and it worked..

closed 2024-06-07 by rvalotta