grblHAL/core #527: Syntax error / typo in ngc_expr.c

MitchBradley, 2024-06-01:
Line 794 of ngc_expr.c is:
```
                if((stack_index > 1) && precedence(operators[stack_index - 1] <= precedence(operators[stack_index - 2])))
```
I think there is a misplaced closing parenthesis.  It should be
```
                if((stack_index > 1) && precedence(operators[stack_index - 1]) <= precedence(operators[stack_index - 2]))
```

terjeio, 2024-06-02:
Good catch!
I (or someone else?) should write a comprehensive test suite for the parameters/expressions/flow control part of the core...

MitchBradley, 2024-06-02:
Thank the C++ compiler for catching it.  I tried compiling with C++ and the stricter type coercion revealed the problem.

I think there might also be an issue with EXISTS[].  [LinuxCNC](https://linuxcnc.org/docs/html/gcode/overview.html#gcode:functions) says that the syntax is EXISTS[#<_name>], but as I read the code, it is looking for e.g. EXISTS[_name] without the enclosing #<>.

terjeio, 2024-06-03:
> I think there might also be an issue with EXISTS[].

You are correct, I will fix this in the next commit.