grblHAL/core #535: $150 - $153 does not work on SKR Pico

JivaroFAD, 2024-06-15:
When I change axis microstepping value on SKR Pico (for all 4 axis), the board respond "ok" but does not memorize the new parameter and stay to the default value (16).

I use a firmware compiled by the web builder.

$I
[VER:1.1f.20240604:]
[OPT:VNDSL,100,1024,4,0]
[AXS:4:XYZA]
[NEWOPT:ENUMS,RT+,ES,SED,RTC]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:216K]
[DRIVER:RP2040]
[DRIVER VERSION:240408]
[DRIVER OPTIONS:SDK_1.5.0]
[BOARD:BTT SKR Pico 1.0]
[AUX IO:0,1,0,0]
[PLUGIN:Trinamic v0.16]

terjeio, 2024-06-20:
Is the motors/drivers powered when you attempt to set the microstepping?
If not powered when the controller tries to initialize them initialization fails and the settings will not stick.

JivaroFAD, 2024-06-21:
Thanks for your response :)
Ok, I've found : the drivers were powered, but the "usb" cap was still in place. After removing it and rebooting board, all works fine.

closed 2024-06-21 by JivaroFAD