grblHAL/core #538: Typo in settings.c

andrewmarles, 2024-06-21:
I think that on lines 2010 and lines 2023 that the bang should not be prefixed on the call to spindle_get_caps:
https://github.com/grblHAL/core/blob/9a060d7ea0f01cab7444fee447741b010cb4cb3a/settings.c#L2010
https://github.com/grblHAL/core/blob/9a060d7ea0f01cab7444fee447741b010cb4cb3a/settings.c#L2023

I think they should be 

`available = hal.signals_cap.safety_door_ajar && spindle_get_count() && spindle_get_caps(true).at_speed;`

Otherwise the spindles with speed feedback all return false on this check.

terjeio, 2024-06-22:
This gets complicated when more than one spindle (not counting spindles that are not spinning) is configured and not all are at speed capable since the value is used as the timeout for at speed capable spindles. I guess a new setting for the timeout is required.

terjeio, 2025-02-17:
No longer relevant due to recent settings changes. Closed.

closed 2025-02-17 by terjeio