grblHAL/core #542: Request for assigned settings for toolsetter moves

LoganFraser, 2024-07-02:
I have a workflow request that would make probing and toolsetter use easier. Currently I have my G59.3 set to 60mm above my toolsetter for the longest to shortest tool stickout + some clearance. If I set probe distance to 65mm it works fine for toolsetting but often when probing parts on Z axis I have to reduce back to 10mm to avoid getting an out of travel error. 

The request - add a seperate value for the toolsetter distance that's used for TLR/TLO only.

terjeio, 2024-07-04:
> to avoid getting an out of travel error.

Soft limits get triggered? If so would it help if I limit the probing move to be within the soft limits?

LoganFraser, 2024-07-15:
Ah, yes.  That would certainly solve it.

closed 2024-09-25 by terjeio