grblHAL/core #549: Teensy 4.1 Mysterious dropped or added characters in gcode over USB

clashoftornados, 2024-07-15:
I have been having an issue that I cannot seem to isolate with sending GCode to Teensy 4.1 over both IOSenderXL and UGS, on 3 separate computers over multiple types of USB cables, both Mac and Windows environments. 
I added some extra debug code into gcode.c to print out what data was received over USB serial, and I am consistently seeing it either omit the G, or send 2 Gs in random Gcode lines. I had first suspected EMI issues so had twisted and shielded all connections to the main T41U5XBB control board. Nothing seemed to work, so I tried running this same test on a bare teensy 4.1 board with nothing connected, and tried several usb cables. I even tried using an external power supply instead of the USB, and the failure always occurs before reaching the end of the file.
I also tried with PJRC library and Arduino USB, as well as using PIO and Arduino IDE to upload the code. I also tried setting the CHECK_MODE_DELAY to a higher value (20) and dropping the BAUD rate to 115200 instead of 230400. Nothing I do seems to make any difference and I am at a loss as to what to do next, so I am making an issue here.

I have also tried a second Teensy 4.1 board which also shows the issues. 
Maybe related, my Z axis on my T41U5XBB seems to not function (though the IO on my teensy on scope shows it working correctly and generating the pulses), I have just set it to use the A axis instead as this is a 3 axis machine.
Default probe pins seems to not function either, though they work on the Teensy side when shorting out the GPIO.

as an example of debug output:
Failed to run GCODE error 1. recv: 1X18.3480Y18.0829

This is the gcode file I am trying to send, generated by kiri:moto
[small-sun-1-16.zip](https://github.com/user-attachments/files/16241166/small-sun-1-16.zip)

my_machine.h and config.h:
[config.zip](https://github.com/user-attachments/files/16241184/config.zip)

Commit : iMXRT1062-master 68c18be2853f99bc70dafd9a4490d4c43c09e696

I would very much appreciate some input or help on what I could try next, or how I could go about further debugging this issue.

terjeio, 2024-07-21:
Have you tried with a different Teensy board? Or with an ethernet connection?

clashoftornados, 2024-07-21:
Yes, I have tried with 2 different Teensy 4.1 boards, both bare boards without anything else attached and the proper axis inversions so it doesn't alarm immediately. The results are the same with it eventually failing somewhere along the way. I don't have the Ethernet hardware for it as of now and that would be very inconvenient given where my network equipment and workshop are located. I really wanted to be able to use USB control for this so I could have a laptop attached to the machine and act as wireless control.

terjeio, 2024-07-21:
Have you tried with a [Web Builder](http://svn.io-engineering.com:8080/?driver=iMXRT1062&board=T41U5XBB) build?
Asking since I have no idea why USB fails for you except due to EMI, a faulty board or perhaps a framework issue.
I have run your gcode numerous times without issues, and there are 1000+ Teensy based grblHAL boards out in the wild and only you have reported this problem. So the grblHAL codebase is likely ok.

clashoftornados, 2024-07-23:
Hmmm, I tried the web builder and that doesn't crash, so something with my build environment or something is clearly messed up. Thanks for your help

closed 2024-07-23 by clashoftornados

terjeio, 2024-07-24:
Which version of Teensyduino?

clashoftornados, 2024-07-26:
Teensy 1.5.9 / IDE 2.3.2. I know I saw something about the latest being broken as of 2022, but it's kind of surprising the latest is still broken almost 2 years later, if that is the problem. Is there anything that can actually be done to fix that, or is it a bug in the actual teensy framework? I don't currently have any JTAG debuggers of my own to actually dig deeper into the root cause.

terjeio, 2024-07-30:
I am using 1.56/1.8.13. The eariler USB issue was due to a bug in the framework (or a compile/linker issue), perhaps it has resurfaced?
Anyway, I am not prepared to spend countless hours trying to debug this again when the older version just works.