grblHAL/core #550: Controllo motore Nidec con mks dlc32 e CLK (no PWM)

distebia, 2024-07-18:
Hi, I have a 6 wire Nidec brushless motor with 24v-Gnd-5V-CLK-BRK-dir. I tried to control it with the MKS DLC32 and I can move it clockwise and anticlockwise using m3 and m4. I can adjust the speed only by varying the pwm value from the firmware, if I set it to pwm 500 it runs slow and jerky, if I set it to 20000 it runs fast and fluidly. Any value sent from consoles like m3 s1 (or s1000) does not change the speed. Is there a way to generate the CLK signal instead of pwm?

terjeio, 2024-07-18:
Link to datasheet please.
If controlled by frequency a custom spindle driver has to be added.

distebia, 2024-07-18:
I don't have a spec sheet for the motor, what I know is that it is labeled nidec g1271170. from the first tests I can operate it with m3 forward and m4 backward and I vary the speed only by changing the pwm hz $33 value in the firmware. any command from m3s1g1f1000 to m3s1000g1f1000 does not change the speed.

terjeio, 2024-07-18:
> ...I vary the speed only by changing the pwm hz

Well, then I assume it controlled by frequency, not duty cycle or DC. If so a custom spindle driver has to be written.

distebia, 2024-07-19:
ma il driver mandrino personalizzato può essere applicato a esp32 mks dlc32?

distebia, 2024-07-20:
[pdfcoffee.com_nidec-24h-bl4234-4-pdf-free.pdf](https://github.com/user-attachments/files/16322401/pdfcoffee.com_nidec-24h-bl4234-4-pdf-free.pdf)
here is the technical data sheet, the model is 27h

terjeio, 2024-07-21:
There are two variants according to the datasheet, PWM or PLL controlled. Since it looks like you have the PLL version a custom spindle driver has to be written.

distebia, 2024-07-22:
#include "driver.h"
#include "grbl/plugins.h"

// Definisci il pin GPIO per l'uscita della frequenza
#define SPINDLE_FREQ_PIN 1032

// Funzione per inizializzare il plugin
static void spindle_freq_init(void) {
    // Configura il pin GPIO come uscita
    hal.gpio.pin_mode(SPINDLE_FREQ_PIN, GPIO_MODE_OUTPUT);
}

// Funzione per aggiornare la frequenza del mandrino
static void spindle_freq_update(uint_fast16_t pwm_value) {
    // Converti il valore PWM in una frequenza (esempio semplice)
    uint32_t frequency = pwm_value * 100; // Modifica questa formula secondo le tue necessità

    // Imposta la frequenza sul pin GPIO
    hal.gpio.write(SPINDLE_FREQ_PIN, frequency);
}

// Funzione di callback per il cambio di velocità del mandrino
static void spindle_set_state(spindle_state_t state, float rpm) {
    uint_fast16_t pwm_value = rpm_to_pwm(rpm); // Converti RPM in valore PWM
    spindle_freq_update(pwm_value);
}

// Struttura del plugin
static plugin_t spindle_freq_plugin = {
    .name = "Spindle Frequency Plugin",
    .version = "1.0",
    .init = spindle_freq_init,
    .spindle_set_state = spindle_set_state
};

// Funzione per registrare il plugin
void register_spindle_freq_plugin(void) {
    plugin_register(&spindle_freq_plugin);
}

I don't know how to write a plugin..I asked bing and I don't know if the result is correct...can I go ahead with the plugin or am I completely off track?

terjeio, 2024-07-22:
I am afraid that it is a bit more complicated.
I think the best option is to use a PWM peripheral and change the frequency instead of the pulse width on RPM changes.
Currently the [LED peripheral](https://github.com/grblHAL/ESP32/blob/a5aef32ab2f3f4f9a66354072c03ac981994f98e/main/driver.c#L1746-L1785) is used for the PWM spindle, not sure if this is suitable. One has to consult the datasheet to find out.

distebia, 2024-07-22:
so the plugin generated by bing is incorrect?

distebia, 2024-07-23:
if bing generates an incorrect plugin for me I don't know how to do it, I don't know how to write it, I'm completely naked in the sector. Besides, I can't even integrate it for compilation

terjeio, 2024-07-24:
The generated code is unusable.
The easiest option for you is to buy the PWM version, or you will have to wait until I get around to write driver code. This may or may not happen since this is the first time I have heard of a single phase frequency controlled motor.

distebia, 2024-07-25:
I managed to start the spindle, it's not the best as the speed doesn't change during processing but I still get the expected result... I create a macro with -$33=200- , -m3s1g1f1000-, I create another one with -$33=5000- , -m4s1g1f1000...from the combination of the values ​​I can go forward and backward at different speeds, thanks for the support..
I saw that the stepper spindle for mks dlc32 is supported...how should I use it?

closed 2024-11-20 by distebia