grblHAL/core #551: When homing is ongoing real-time report requests are ignored？

attying, 2024-07-20:
hello terjeio，I found that sending '?' does not receive any response When homing is ongoing. This is somewhat different from FluidNC. Can real-time reporting be enabled, or where should I modify it in the source code?

terjeio, 2024-07-21:
> I found that sending '?' does not receive any response When homing is ongoing.

Correct, this is how legacy grbl behaves.

> Can real-time reporting be enabled, or where should I modify it in the source code?

[It can](https://github.com/grblHAL/core/blob/8b9b7859ee6b370442d380d3d754fa5392b064b2/machine_limits.c#L216-L245) but should it? Any reason for why?

attying, 2024-07-21:
I am glad to receive your prompt reply. I hope to observe the changes in the machine position during the process of returning home, which can be useful in certain situations. For example, during debugging or when the machine travel is relatively large. FluidNC and Mach3 are like this, can you tell me where to enable or modify them? I really need it, thank you!

attying, 2024-07-21:
i 'm sorry! I didn't notice your mark. The problem has been resolved, thank you very much for your answer!

closed 2024-07-21 by attying