grblHAL/core #580: Breaks in received message

karoria, 2024-09-06:
Hi @terjeio 
I don't know where to put this issue in core or iMXRT1062. I get long messages in parts from controller. This makes fetching required data hard. I am not sure whether it has anything to do with firmware code or web sockets driver or browser settings. Please guide me. Screenshot of chrome dev tools console is attached here.
Thanks in advance.
Ravi
[image]

karoria, 2024-09-07:
The same behaviour is seen in $$ response also.

terjeio, 2024-09-07:
TCP/IP is a packet based protocol and messages might thus be sent in chunks. In addition there is buffering in grblHAL and to improve troughput without adding too much delay messages are sent periodically without waiting for the output buffer to fill completely.
So not much can be done at the transmitting end, you will have to parse the incoming data at the receiving end. One option is to wait for the line terminating characters before proceeding, e.g. in ioSender I do so and fire an event for each line received so that the parsing code always gets a complete line at a time.

karoria, 2024-09-07:
Got it. Thanks.

closed 2024-09-25 by terjeio