grblHAL/core #588: Setting $14 to any value freezes STM32F103RC controller

macbef, 2024-09-26:
I'm running into a new issue with my custom board again.

It seems that everything is working fine when I use a version I have built last year. 
After updating to the most recent commit, my board seems to be working correctly (responding to commands etc) and I can set all parameters without problem.

**Except for $14**. When I try to change $14 (and even with just sending the active value again, like $14=0), the MCU hangs. And the value is not stored as well.

I guess it's not a hardware issue because the older version is working fine on my board.

Any suggestions? Maybe related to the EEPROM-emulation? (I'm mot using an external EEPROM)

Thanks!

Loewe111, 2024-09-27:
I have the same issue on a RP2040

macbef, 2024-09-27:
Anything I can do to debug this?
I have compiled this for USB or UART, the error occurs for both.

When I revert to build 20230217, everything seems to be fine.

terjeio, 2024-09-27:
There has been a few changes to the handling of input pins so it is hard for me to tell what is wrong from just looking at the code - I am abroad for a while and do not have a board to debug with.

> Maybe related to the EEPROM-emulation? (I'm mot using an external EEPROM)

The emulation is in the core, and I do not have the same issue with the boards I has access to. So most likely the bug is located in the driver.

> Anything I can do to debug this?

Do you program with a debug probe and are able to halt execution and/or set breakpoints?
If you can halt where does it stop? In the hardfault handler?

@Loewe111 Which board do you have and which version of grblHAL ($I output). FYI I do have a RP2040 with me and that works.

Loewe111, 2024-09-27:
The Board i use is the Seed Xiao RP2040 (https://www.seeedstudio.com/XIAO-RP2040-v1-0-p-5026.html)
$I is 
```
[VER:1.1f.20240921:grblHAL on Blot]
[OPT:VNCDSL,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,NOPROBE,ES,SED,RTC]
[FIRMWARE:grblHAL]
[SIGNALS:HSE]
[NVS STORAGE:*FLASH]
[FREE MEMORY:220K]
[DRIVER:RP2040]
[DRIVER VERSION:240907]
[DRIVER OPTIONS:SDK_1.5.1]
[KINEMATICS:CoreXY v2.01]
```

terjeio, 2024-09-28:
For both of you: is the setting lost on a power cycle? or is the board still unresponsive so you cannot check?

@Loewe111 I have tested with my dev board and both a local and a Web Builder build and I am not able to replicate the issue.

Loewe111, 2024-09-28:
The settings gets reset to 0 for me
Im gonna try to run on a normal pi pico and then debug later

Loewe111, 2024-09-28:
The Normal Pico does the same for me. 
I goes into a `isr_hardfault` in `void ioport_setting_changed (setting_id_t id)` in `ioports.c`
Callstack:
´´´
isr_hardfault@0x100001c4 (Unknown Source:0)
<signal handler called>@0xfffffff9 (Unknown Source:0)
ioport_setting_changed@0x1001bbfc (Unknown Source:0)
ioport_setting_changed@0x1001bbca (Unknown Source:0)
set_control_invert@0x10013790 (Unknown Source:0)
settings_store_setting@0x10014d12 (Unknown Source:0)
system_execute_line@0x10019848 (Unknown Source:0)
protocol_main_loop@0x1000eda8 (Unknown Source:0)
grbl_enter@0x100046a4 (Unknown Source:0)
main@0x1000030e (Unknown Source:0)
´´´

Couldn't figure out anything more tho

terjeio, 2024-09-28:
> I goes into a isr_hardfault in void ioport_setting_changed (setting_id_t id) in ioports.c

That may be helpful. If there are no input pins defined it will likely cause a hardfault.

Can you try changing [this line](https://github.com/grblHAL/core/blob/5d6a99c5839737f5fb855c16ac4c2369ae37b66f/ioports.c#L789) to:
`if(port) do {`
and try again?

terjeio, 2024-09-28:
or even better[change this](https://github.com/grblHAL/core/blob/5d6a99c5839737f5fb855c16ac4c2369ae37b66f/ioports.c#L752) to:

`    else if(digital.in.ports && digital.in.ports->n_ports) switch(id) {`

macbef, 2024-09-28:
YES! This solves this issue.

Could you tell me which input pin definition might be missing? I have a probe pin defined in my machine-map.

// Define probe switch input pin.
#define PROBE_PORT              GPIOC
#define PROBE_PIN               3

Some other inputs are undefined (I2C_STROBE and SDCARD stuff that I don't need)

Anyways, always a pleasure to see how fast you are reacting to issues like this one!
Thanks a lot

macbef, 2024-09-28:
Issue solved

closed 2024-09-28 by macbef

Loewe111, 2024-09-28:
that fixes it for me aswell

macbef, 2024-09-28:
This fix was essential for my new board. I always wanted to have a board with opto-isolated USB and opto-isolated RS232 interfaces. When driving bigger motors, soon or later you'll run into ground-loop problems.
Now this one is working.
@terjeio , let me know if you want one. It only has 3 axes and no drivers on board, though. Just pinheaders for MKS TMC2160-OC and alike.