grblHAL/core #604: SKR PRO 1.2 TMC5160 X axis not working

ggc-r, 2024-10-09:
Hello,
I'm having issues with the build 20241006. I used the web builder to compile an image for the SKR PRO 1.2 board. When I tested the controller the X axis wasn't working. I had no issues with YZA axes. Next, I swapped the cables for the X and Y motors and now the Y motor did not work. This suggests that the X motor's cable is not at fault. Next, I tried swapping the motor driver with a known working one from Y axis to the X-axis and this also did not fix the issue.

terjeio, 2024-10-10:
From the status report (ola, olb) it looks like the motor is not connected, but this is not a reliable way to determine that. When you try to move X is the motor engaged and cannot easily be turned? And if you command a X-move and send M122X while the move is ongoing is ola and olb still flagged with *?

ggc-r, 2024-10-10:
I connected the X-axis motor to the Y-axis stepstick to test the electrical connection and the motor worked fine. I also moved a known working Step Stick to the X-axis, which did not help. The X-axis step stick also does not produce a holding current(motor does not hold the position). This is the output of M122 X when a move is ongoing. 
> M122 X
[TRINAMIC]
                      X
Driver          TMC5160
Set current         999
RMS current         990
Peak current       1400
Run current       19/31
Hold current       9/31
Global scaler   133/256
CS actual         19/31
PWM scale            26
vsense              N/A
stealthChop       false
msteps               16
tstep               806
pwm
threshold             0
[mm/s]                0
OT prewarn        false
OT prewarn has
been triggered    false
off time              5
blank time            1
hysteresis
-end                  2
-start                6
Stallguard thrs       0
DRIVER STATUS:
stallguard
sg_result             0
fsactive
stst
olb                   *
ola                   *
s2gb
s2ga
otpw
ot
STATUS REGISTERS:
 X = 0x60:13:00:00
ok
>
I am convinced that, the issue is either with the SKR PRO 1.2 PCB or the software.
I tried using a multimeter on the outputs of the stepstick and the resistance of the motor phases was nominal.
I think as a next step I will try to upload different releases of GRBLhal to check for software issues. I really don't want to pull out my logic analyzer just yet.

terjeio, 2024-10-11:
What is your $4 (stepper enable invert) setting value? And can you check the enable signal on the X stepstick?

ggc-r, 2024-10-11:
My $4 is set to 15
> $4
$4=15 (Invert step enable pin, boolean)
ok
I measured the state on each driver enable pin, and the X-axis enable pin was high while others were held low.

terjeio, 2024-10-11:
My bad - a recent typo/regression, sorry about that. Should be ok now, fix committed.

ggc-r, 2024-10-14:
I just checked at the X-axis is working fine now. Thank you for your help.

closed 2024-10-14 by ggc-r