grblHAL/core #609: Trying to home a second time causes a crash

ramin-afshar, 2024-10-16:
I flashed the BB 32 firmware with a build generated through [http://svn.io-engineering.com:8080/?driver=ESP32&board=BlackBox X32](http://svn.io-engineering.com:8080/?driver=ESP32&board=BlackBox%20X32) . When I start my machine and press the Home All button in the Openbuilds Control software my machine is homed and squared as expected. Everything else, like starting a job or jogging the machine works as well. However, if I press the Home All button again I get the following error and crash.

Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC : 0x400dafb0 PS : 0x00060330 A0 : 0x800d8e5f A1 : 0x3ffb63a0
A2 : 0x3ffb643c A3 : 0x3ffb6410 A4 : 0x3ffb6804 A5 : 0x3ffc3718
A6 : 0x00000000 A7 : 0xc42feb33 A8 : 0x3ffb63ac A9 : 0x3ffb63ac
A10 : 0x00000000 A11 : 0x3ffb6804 A12 : 0x00000014 A13 : 0x3ffc36dc
A14 : 0x000124f8 A15 : 0xc42feb33 SAR : 0x00000007 EXCCAUSE: 0x0000001c
EXCVADDR: 0xc42feb47 LBEG : 0x4000c2e0 LEND : 0x4000c2f6 LCOUNT : 0x00000000
Backtrace:0x400dafad:0x3ffb63a0 0x400d8e5c:0x3ffb6410 0x400d91b3:0x3ffb64c0 0x400da511:0x3ffb64f0 0x400e3e2a:0x3ffb6530 0x400e3eb9:0x3ffb6550 0x400e4049:0x3ffb6570 0x400dbe3d:0x3ffb65b0 0x400d45f8:0x3ffb65f0 0x400d1e7f:0x3ffb6620
ELF file SHA256: bdea617fefcb93fd
CPU halted.

I simply disconnect and reconnect and I can then home again.

avignatenko, 2024-10-22:
I havethe same issue on ESP32 (makerbase dlc32 2.1 board).

avignatenko, 2024-10-23:
> @avignatenko Which version (build date)?

2024-10-12.

Should I try updating to the latest one from the web builder?

terjeio, 2024-10-23:
> Should I try updating to the latest one from the web builder?

[Yes](https://github.com/grblHAL/core/commit/236a40c10073ecbf01cead70df1d4d630e758458)

closed 2024-10-24 by terjeio